s-allius/tsun-gen3-proxy #1: Dokumentation how to loop the proxy into the connection

s-allius, 2023-09-25:
It's unclear how the proxy can be looped into the connection between the inverter and the TSUN cloud.

Some examples would help

morkmorks, 2024-04-02:
Indeed- Is there an easy way of setting the single DNS entry in the settings of the commonly used FRITZ!Box (w/o using Pihole)?

s-allius, 2024-04-02:
I'm sorry, but my knowledge is, that you can't add DNS entries on a Fritz!Box directly. So you must use a local DNS Server, like Phiole, for that.
If you have a newer GEN3Plus Inverter, you can configure the connection to the proxy on the Web-Interface of the inverter. The newer inverters have a serial number starting with 'Y17'. Which kind of inverter do you have?

morkmorks, 2024-04-03:
Hi Stefan

 

Thank you so much for the speedy response.

I have an R17…..-inverter – I bought it last year, so no web-Interface.

 

I will try on the Pihole on the weekend – let’s see!

 

Cool that you care about this stuff!

 

Cheers

 

mm

ryanrudak, 2024-05-26:
I did it like that way:
 
	 - enter the tsun-gen3-proxy (IP address or DNS name) as the optional server on the /config_hide.html page of the microinverter
	 - Internet access for the IP address of the mircoinverter is prohibited in the Fritz!Box
 
Since the first server is then no longer accessible, the optional server is used, in this case the tsun-gen3-proxy.
 
This works great for my.

jo-mueller, 2025-06-01:
Late to the party but I think I'm struggling at the same step.

> If you have a newer GEN3Plus Inverter, you can configure the connection to the proxy on the Web-Interface of the inverter. The newer inverters have a serial number starting with 'Y17'. Which kind of inverter do you have?

Mine starts with Y47 which I believe is indicative of it being of a newer generation. As an interface, I use the TSUN smart app - is that what's meant by the web interface? 🤔