s-allius/tsun-gen3-proxy #25: data_type: 76 not supported

limes007, 2023-11-04:
Thank you for sharing your TSUN proxy with us.

In the log file I see the following error

`ERROR | root | Infos.parse: data_type: 76 not supported`

Is this the intended behavior or do I need to change some setting to get rid of this error?

s-allius, 2023-11-04:
No, this is not expected behavior. This seems to be coding that I have not seen before.

Does this occur regularly? 

In the trace.log file should be the hex-dump of the package. Simply use the timestamp to determine the package. 
If you would send me the file or an excerpt, I can take a look at what it could be.

I don't know if you can send a private note via GitHub. We certainly shouldn't publish trace files here

limes007, 2023-11-05:
I get this error every 5 minutes with each transmission of the inverter.
It seems, that I have a newer firmware (RSW_400_V1.00.20) than most other users, this may be the reason you never saw this.

I also don't know how to send private messages here. Is it possible to anonymize the trace somehow or can I publish an excerpt?

s-allius, 2023-11-05:
have set up an email account for this issue. You can send your traces to DataType76@allius.de. As soon as we have a solution, I will delete this email account again. 
For the future, I need to think about how to support an anonymous trace. But that will take too much time for your current problem

limes007, 2023-11-06:
@s-allius You've Got Mail

s-allius, 2023-11-06:
I added support for parsing of int64 data type to info.py. Can you check if this solve this issue

limes007, 2023-11-07:
Thank you. The error is gone now.

closed 2023-11-07 by limes007