s-allius/tsun-gen3-proxy #36: Since Home Assistant Core Update and TSUN Update no Data Transfer

ninjadriver, 2024-03-09:
Aktuell funktioniert bei mir die Übertragung nicht mehr ordnungsgemäß.
Sowohl nach TSUN noch an Home Assistant.
Im MQTT sehe ich, dass die Config übertragen wird, jedoch die Werte werden nicht übermittelt.
Im Proxy Home Assistant Eintrag wird ein Internal SW Error hoch gezählt und die Verbundenen Controller zählen auf 300 und höher nach Oben.

closed 2024-03-09 by ninjadriver

s-allius, 2024-03-10:
Hmm, seit heute Morgen habe ich das gleiche Problem. Der Wechselrichter meldet sich nicht mehr im WLAN an und somit bekommt der Proxy keine Daten.
Ein neues Paring per Bluetooth funktioniert derzeit auch nicht. Evtl. habe ich auch ein Update bekommen, ich gehe mal auf die Suche...

ninjadriver, 2024-03-10:
Ich hatte in der Tat auch das WLAN gewechselt, aber eher da ich einen Router getauscht habe.
=> Ich habe gestern noch festgestellt dass bei mir wohl ein Docker Update auch noch mit reingespielt hat. Wodurch ich den Eintrag auch erst einmal bis zur genaueren Analyse geclosed hatte.

Bei mir also folgendes Verhalten:
- Der Proxy Count ist immer sobald die WR (2) aktiv waren auf 300 und mehr hoch gegangen.
- Nach eingehender Analyse habe ich festgestellt, dass der nslookup im container logger.talent-monitoring.com auf die docker server ip auflöst. (Somit endlos schleife produziert)
- dadurch müssen sich i.wie die Informationen bis auf den Converter (WLAN etc. hat immer funktioniert) überschrieben haben.
- Also habe ich meinen docker container mit 2 DNS Server (google und cloudflare) neu deployed.
- Heute scheint es nun wieder zu funktionieren. (Ich bin gespannt wie lange :P)

Leider hatte das nun wieder funktioneren, den Effekt dass die History DB von Home Assistant (SQLLite) zerbröselt ist.
Schuld waren die Informationen der PV Module. Dort wurde eine execption erzeugt und dadurch die DB verschoben und mit einer leeren gestartet.

s-allius, 2024-03-10:
Erstmal gut wenn es wieder läuft. Wenn der Container die URL auf eine lokale IP oder die eigene IP auflöst, könnte ich ja erkennen und eine sinnvolle Meldung ausgeben und die Verbindung abklemmen, um die Loop zu unterbrechen. Dann sollte zumindest HA noch Daten bekommen. 
Ich schreibe das mal auf meine Ideenliste. Ich schließe den Bug jetzt erstmal wieder. Wenn du neue Infos, Analysen hast, dann können wir ihn ja nochmal öffnen

Gruß Stefan