s-allius/tsun-gen3-proxy #46: Problem to start the container

datenangebot, 2024-04-08:
Hi,

first, thank you for this amazing promising project!
I am looking forward to use this in my home setup...

Unfortunately I am not able to start the container as you described the startup process.
I build the image and tried to start the container by running:
`docker run  --dns '192.168.212.20' --env 'UID=1000' -p '5005:5005' -p '10000:10000' -v /srv/tsun-proxy/config:/home/tsun-proxy/config -v /srv/tsun-proxy/log:/home/tsun-proxy/log tsun-proxy`

And I get the following:
```
######################################################
# prepare: 'tsun-proxy' Version:
# for running with UserID:1000, GroupID:1000
#
# create user
######################################################
#
'2024-04-08 20:20:23  INFO | root | Server "tsun-proxy - " will be started'
'2024-04-08 20:20:23  INFO | root | Initialize proxy statistics'
Traceback (most recent call last):
  File "/home/tsun-proxy/./server.py", line 83, in <module>
    Inverter.class_init()
  File "/home/tsun-proxy/inverter.py", line 21, in class_init
    cls.entity_prfx = ha['entity_prefix'] + '/'
                      ~~^^^^^^^^^^^^^^^^^
KeyError: 'entity_prefix'
```

Is this an error in the code or on my side?

Background:
I tried to use a lxc container and a debian vm under proxmox virtualization.

Btw: Do you have a donation option?

s-allius, 2024-04-08:
Hi datenamgebot, 
it seems to me, that there is something missing in the `config.toml` file. Have you build one? You can start with renaming the `config.example.toml` into `config.toml`

s-allius, 2024-04-08:
In version 0.7.0 we want a helpful message instead of the python crash

datenangebot, 2024-04-09:
Thanks for the hint.
Yes, that was a simple solution.
For me it wasn't that clear, that you have to copy and adapt the config file even if you try tu run the default values...

s-allius, 2024-04-09:
Ok, thanks for your feedback. I will look into how I can improve the initial setup

s-allius, 2024-04-10:
fixed by #48

PR #48 (2024-04-10): print helpful messages on config errors
fix #46
closed 2024-04-10 by s-allius