s-allius/tsun-gen3-proxy #71: Auto Discovery does not connect

Pummelfisch, 2024-05-14:
Sorry if I miss something obvious. I'm new to this tool and Home Assistant in general.
# Problem
With my current configuration I am not able to find any hint of the proxy or the inverter (MS800) in Home Assistant, using docker.
# Setup
I've deployed the stack according to the provided compose file [here](https://github.com/s-allius/tsun-gen3-proxy/blob/main/docker-compose.yaml). I've adjusted the network name and the volumes, but nothing major.
All three containers do start and dont report any error. The proxy is able to connect to mosquitto. I tried my best to configure the config.toml file:

```
# configuration to reach tsun cloud
tsun.enabled = true   # false: disables connecting to the tsun cloud, and avoids updates
tsun.host    = 'logger.talent-monitoring.com'
tsun.port    = 5005

# mqtt broker configuration
mqtt.host    = 'Mosquitto'   # URL or IP address of the mqtt broker
mqtt.port    = 1883
mqtt.user    = 'secure_username'
mqtt.passwd  = 'secure_password'

# home-assistant
ha.auto_conf_prefix = 'homeassistant'       # MQTT prefix for subscribing for homeassistant status updates
ha.discovery_prefix = 'homeassistant'       # MQTT prefix for discovery topic
ha.entity_prefix    = 'tsun'                # MQTT topic prefix for publishing inverter values
ha.proxy_node_id    = 'proxy'               # MQTT node id, for the proxy_node_id
ha.proxy_unique_id  = 'P170000000000001'    # MQTT unique id, to identify a proxy instance

# microinverters
inverters.allow_all = true   # allow inverters, even if we have no inverter mapping

# inverter mapping, maps a `serial_no* to a `mqtt_id` and defines an optional `suggested_place` for `home-assistant`
#
# for each inverter add a block starting with [inverters."<16-digit serial numbeer>"]
[inverters."inverter_id"]
node_id = 'wechselrichter'                 # Optional, MQTT replacement for inverters serial number
suggested_area = 'balkon'          # Optional, suggested installation area for home-assistant
pv1 = {type = 'PiE AIR superLIGHT Half-Cut 200+', manufacturer = 'PiE'}   # Optional, PV module descr
pv2 = {type = 'PiE AIR superLIGHT Half-Cut 200+', manufacturer = 'PiE'}   # Optional, PV module descr
```
> I've removed any sensitive data like username and password aswell as the inverter id.

Also I have configured mosquitto with a config file, activating `per_listener_settings` and providing a `password_file`.

In the Home Assistant Web Interface I cant connect to the Proxy. Am I missing something or missconfigured anything?

Thank you in advance for the tool in general and any advise. If you need any furthor information, please let me know.

limes007, 2024-05-14:
With one of the recent updates I had to add some configuration settings for newer inverters, even though I don't have them.

Maybe you need this settings too.

```
# configuration to reach the new tsun cloud for G3 Plus inverters
solarman.enabled = false   # false: disables connecting to the tsun cloud, and avoids updates
solarman.host    = 'iot.talent-monitoring.com'
solarman.port    = 10000
```

Pummelfisch, 2024-05-14:
Thank you for the idea. Actually I just removed those from the file, since i thought those where not necessary for me. Just re-added those lines but nothing changed.

s-allius, 2024-05-14:
You can easily test the connection between Home Assistant and your proxy. 
Restart you Home-Assistant and you should see these lines in your proxy docker console:
```
'2024-05-14 21:50:14  INFO | mqtt | Home-Assistant Status: offline'
'2024-05-14 21:50:33  INFO | mqtt | Home-Assistant Status: online'
'2024-05-14 21:50:33  INFO | root | Initialize proxy device on home assistant'
```
That's because the proxy also listen on MQTT packets from home assistant. The third lines say, that the `auto configuration` was started.
If you don't see this lines you have a problem with you MQTT configuration in HA or on your proxy...

Pummelfisch, 2024-05-15:
I was able to add an mqtt instance by hand under the integrations page of home assistant. I assume that this is actually what i missed. Prior i just assumed that auto detect means that home assistant kind of suggests the proxy as a possibility. My missunderstanding sorry for that.
Now I am able to se these exact three lines in in the proxy container.

Could you also tell me where i would be able to see if the inverter is connected succesfully to the proxy, because at the moment there is no output in Home Assistant that i can see?

Pummelfisch, 2024-05-15:
Also when I compare your Screenshot of Devices in HomeAssistant [here](https://private-user-images.githubusercontent.com/122395479/275958193-f5e5230a-b685-4984-a9e9-f528c477d130.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU3NjYzOTEsIm5iZiI6MTcxNTc2NjA5MSwicGF0aCI6Ii8xMjIzOTU0NzkvMjc1OTU4MTkzLWY1ZTUyMzBhLWI2ODUtNDk4NC1hOWU5LWY1MjhjNDc3ZDEzMC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxNVQwOTQxMzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lZTEzMTBkYTBkOTk2ZjY4YmUxOWEyYjg1MjljMGZiMjI1Mjg4NDgwNjhlZDFjYTcwOWUyMTVjZDUxMDQyNmFiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.AR1mOEDv0Ng0FZdDlXhstdhJn4rCVKJgR6yT9zABUFs) with what I am able to see there is clearly something missing which makes me think that maybe something else ist not quite correct configured on my side.
[image]

closed 2024-05-15 by Pummelfisch

s-allius, 2024-05-15:
Your are welcome