s-allius/tsun-gen3-proxy #75: Home Assistant mqtt addon does not allow publishing

allgoewer, 2024-05-22:
Hi,

i setup the proxy container as described which is working, i can see inverter messages in proxy.log.

Mu home assistant is ruinming as a separate instance in a vm.

In home assistant i added a user mqtt-tsun-proxy.

I cam see the proxy connect successfully, so username and password match but mosquitto denies all publishes by the proxy.

Here's an excerpt from the log:

```
2024-05-22 21:21:44: Denied PUBLISH from auto-009DD51C-8423-71EB-0D30-4812DEAF0068 (d0, q0, r0, m0, 'homeassistant/sensor/proxy/unknown_ctrl_P170000000000001/config', ... (400 bytes))
2024-05-22 21:21:44: Denied PUBLISH from auto-009DD51C-8423-71EB-0D30-4812DEAF0068 (d0, q0, r0, m0, 'homeassistant/sensor/proxy/ota_start_cmd_P170000000000001/config', ... (395 bytes))
2024-05-22 21:21:44: Denied PUBLISH from auto-009DD51C-8423-71EB-0D30-4812DEAF0068 (d0, q0, r0, m0, 'homeassistant/sensor/proxy/sw_exception_P170000000000001/config', ... (401 bytes))
2024-05-22 21:21:44: Denied PUBLISH from auto-009DD51C-8423-71EB-0D30-4812DEAF0068 (d0, q0, r0, m0, 'homeassistant/sensor/proxy/inv_msg_fmt_P170000000000001/config', ... (407 bytes))
2024-05-22 21:21:44: Denied PUBLISH from auto-009DD51C-8423-71EB-0D30-4812DEAF0068 (d0, q0, r0, m0, 'homeassistant/sensor/proxy/at_cmd_P170000000000001/config', ... (382 bytes))
2024-05-22 21:21:45: Denied PUBLISH from auto-009DD51C-8423-71EB-0D30-4812DEAF0068 (d0, q0, r0, m0, 'tsun/proxy/proxy', ... (201 bytes))
2024-05-22 21:21:45: Received SUBSCRIBE from auto-009DD51C-8423-71EB-0D30-4812DEAF0068
2024-05-22 21:21:45: 	homeassistant/status (QoS 0)
...
```

My mosquitto config in HA looks like this

```
logins: []
require_certificate: false
certfile: fullchain.pem
keyfile: privkey.pem
customize:
  active: true
  folder: mosquitto
anonymous: false
debug: true
```

Any idea what is wrong?

s-allius, 2024-05-22:
Hello allgoewer,

I never had such problems with mosquitto. I search for you problem and get hints, that there might be a problem with the ACL (access control list). Mosquitto have and very strikt default ACL.
Maybe that is your problem. Pls start reading this: http://www.steves-internet-guide.com/topic-restriction-mosquitto-configuration/

allgoewer, 2024-05-22:
You made the right call.

Turns out I created a ACL for mosquitto at some point in the past.

My Home Assistant installation is ancient, I can not remember creating it.

I'll close this as resolved.

closed 2024-05-22 by allgoewer