s-allius/tsun-gen3-proxy #77: Reduce size of the log files

s-allius, 2024-05-27:
With Version 0.8.0 we write huge log files every day. That's why we send every Minute a Modbus Request and get a reply. At sunset the inverter still have the TCP connection up, but the DSP isn't able to answer the request. In this situation we resend  the request a second time. (and in earlier version up to 3 times).

We don't want to see all the packets in the trace, which are produced by the proxy and we also don't want to see the replies for these. So we should reduce the log level for these Modbus Commands to DEBUG.

Modbus packets which comes from TSUN are still very intresting for us and the are very seldom. So we should trace these ones with INFO level. The same for packet which are caused by MQTT commands.

roethigj, 2024-05-27:
In Addition, when The connection to tsun clouds gets lost I get:
```
OSError: [Errno 113] Host is unreachable
'
'2024-05-10 18:55:15 ERROR | conn | Exception for ('192.168.178.55', 62152):
Traceback (most recent call last):
  File "/home/tsun-proxy/async_stream.py", line 63, in loop
    await self.__async_read()
  File "/home/tsun-proxy/async_stream.py", line 109, in __async_read
    data = await self.reader.read(4096)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 694, in read
    raise self._exception
  File "/home/tsun-proxy/async_stream.py", line 63, in loop
    await self.__async_read()
  File "/home/tsun-proxy/async_stream.py", line 109, in __async_read
    data = await self.reader.read(4096)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 713, in read
    await self._wait_for_data('read')
  File "/usr/local/lib/python3.12/asyncio/streams.py", line 545, in _wait_for_data
    await self._waiter
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1013, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

```
permanently.
This blows the log to 30 GB and more!
Only solution is to switch off the cloud in config.

s-allius, 2024-05-28:
I will catch the exception and then close the connection. That will stop the loop. On my side I never saw this error, which comes from the underlying system.

s-allius, 2024-05-31:
fixed by #79 and #82

closed 2024-05-31 by s-allius