s-allius/tsun-gen3-proxy #84: The Compose file './docker-compose.yaml' is invalid

Lamprecht, 2024-06-03:
I see some errors with the docker-compose.yaml file (checked out from main).
`ERROR: The Compose file './docker-compose.yaml' is invalid because:
networks.outside.ipam.config value Additional properties are not allowed ('gateway', 'ip_range' were unexpected)
networks.outside value Additional properties are not allowed ('name' was unexpected)
services.home-assistant.healthcheck value Additional properties are not allowed ('start_period' was unexpected)`

I think these are incompatibilities with the Version 3 declaration. See this question on [SO](https://stackoverflow.com/questions/69334107/docker-compose-invalid-additional-properties-are-not-allowed)

s-allius, 2024-06-03:
Thanks for the hint. I am running my solution on a Synology NAS with a 
Docker version 20.10.23, build 876964a

where I have no problems. My original idea was to not use the default network so that the containers are isolated from others on the machine. I don't have that much experience with Docker, so this approach might be wrong.

I will look for a better solution

s-allius, 2024-06-03:
Update: On my MAC I only got a warning, that the version will be ignored

```bash
docker -v            
Docker version 26.1.1, build 4cf5afa
docker-compose -v    
Docker Compose version v2.27.0-desktop.2
sallius@stefans-macbook-air tsun % 
```

s-allius, 2024-06-03:
@Lamprecht 

I fixed the warnings and removed the obsolete version definition. With both of my docker-compose versions, there a no more errors.

Is it possible for you to check the latest version from the dev-0.8.0 branch with your `docker-compose` version?

Lamprecht, 2024-06-03:
```
Proxy git:(master) > export PROJECT_DIR=/home/chris/devel/TSUN/Proxy
Proxy git:(master) > docker-compose build                           
ERROR: Invalid interpolation format for "home-assistant" option in service "services": "${PROJECT_DIR:-./}homeassistant/config:/config"
```

Lamprecht, 2024-06-03:
Trying with docker-compose-v2:
```
> /usr/libexec/docker/cli-plugins/docker-compose -v
Docker Compose version 2.20.2+ds1-0ubuntu1~20.04.1
> /usr/libexec/docker/cli-plugins/docker-compose up
... lots of pull messages snipped
network home-assistant declared as external, but could not be found
```

s-allius, 2024-06-03:
You probably want to add a slash after:
`export PROJECT_DIR=/home/chris/devel/TSUN/Proxy`

so your path results to:
`/home/chris/devel/TSUN/Proxy/homeassistant/config`

But that is certainly not the problem. I'll have another look tomorrow

s-allius, 2024-06-04:
Now I removed the network definition completely. 

Can you give the latest version of the dev-0.8.0 branch another try?

Lamprecht, 2024-06-04:
Yes, works now!
Thank you very much!

s-allius, 2024-06-04:
Cool. Thank you very much for your support.
I had never seen the problem before. For the future I now have a check in the build script, so that errors should be noticed directly

closed 2024-06-04 by s-allius