s-allius/tsun-gen3-proxy #91: Add Endpoints for Metrics and Probes

Cardes, 2024-06-14:
Use Case:

- As an Operator i want the orchestrator to handle restarts if the container crashes, therefor i need readiness and liveness checks
- Also as an Operator i want to fetch metrics from the proxy so i can visualize the state and trigger alerts on issues

Design Proposal:
- add a port that exposes /metrics in prometheus format using the [prometheus python client library](https://github.com/prometheus/client_python)
- the same port should expose /-/healthy and /-/ready endpoints for [Kubernetes](https://kubernetes.io/docs/tasks/configure-pod-container/configure-liveness-readiness-startup-probes/) and [docker compose](https://docs.docker.com/compose/compose-file/05-services/#healthcheck)

Based on this [discussion](https://github.com/s-allius/tsun-gen3-proxy/discussions/87#discussioncomment-9767962)

s-allius, 2024-06-14:
Great idea. I also find a async version of the prometheus client library.

I think I will start with the health check...

s-allius, 2024-06-15:
I implemented a first `ready`check. This check looks for a proper config file and that both server (Port:5005 and 10000) are started.
If everything is fine, I return on http://172.16.30.7/-/ready a HTTP code 200 and the text `Is ready. If there is a problem in the config file, I return a HTTP code 503 and the text `Not ready`. It is also possible that the HTTP endpoint isn't available on errors or startup.
Does this behaviour fit for k8s?

---
For the health check, I will evaluate the processing time of the messages. This should actually be quite simple and centralized.

Cardes, 2024-06-16:
This looks good, all http codes between 200 and below 400 are considered success, so 503 will get recognized as failed. 
For startup purpose its common to define an initialDelaySeconds or a failureThreshold, so no answer / timeout works fine as long as the container is starting up.
Example:
```
  livenessProbe:
      httpGet:
        path: /-/healthy
        port: 8127
      initialDelaySeconds: 12 #time before the first probe
      periodSeconds: 20 #time between probes
      timoutSeconds: 2 #delay until no answer is considered a timeout (usefull for high load or low compute environments)
      successThreshold: 1 #how many consecutive probes are needed to assume a healthy probe
      failureThreshold: 2 # how many consecutive failed probes are needed to trigger the unhealty state
      terminationGracePeriodSeconds: 10 #how long to wait between shutdown signal and forced stop of the container
```

The Port could be any number above 1024 that's not used for a common service, so 8127 works fine i think. considering that host network is often used in smaller setups and using an uncommon port helps users to prevent port duplications. In every other environment port mapping/ distinct service ips should prevent any collisions.

Thanks for your effort, let me know if i can help test anything.

Regards,
Sebastian