s-allius/tsun-gen3-proxy #119: GEN3: product version changes (TSOL-MS350)

onkelenno, 2024-07-09:
First of all, thank you very much for your effort. Great work.

For me it works perfect with TSOL-MS350 V6.0.014/V6.0.0E.

I do not have a home automation so I only installed mosquito MQTT and subscribe to the topics tsun/#.
One thing I noticed is, that the version in topic tsun/inverter changes every now and then (V6.0.0E and V6.0.014):

2024-07-09 08:01:31.947 tsun/inverter {"Rated_Power": 400, "No_Inputs": 1, "Product_Name": "Microinv", "Manufacturer": "TSUN", "Version": "V6.0.0E", "Serial_Number": "T17XXXXXXXXXXXXX", "Equipment_Model": "TSOL-MS350", "Max_Designed_Power": 65535}
2024-07-09 08:02:13.538 tsun/inverter {"Rated_Power": 400, "No_Inputs": 1, "Product_Name": "Microinv", "Manufacturer": "TSUN", "Version": "V6.0.014", "Serial_Number": "T17XXXXXXXXXXXXX", "Equipment_Model": "TSOL-MS350", "Max_Designed_Power": 65535}

This happends very, very offen.

Is there something missing when converting the raw data from the inverter? In trace.log I can see that the inverter always sends V6.0.0E.

BTW: it would be cool if the topics total, input and grid would also publish a timestamp so one could use persistant sessions.

Thank you

s-allius, 2024-07-09:
Hi onkelenno,

for the problem with the flapping version number I have a possible fix. Can you check this image from docker.io?

```bash
    image: docker.io/sallius/tsun-gen3-proxy:dev
```

Background:
On GEN3 inverters we get the version in two different messages. In the first it is coded as a string and in the second (Modbus response) it is coded as a BCD number. The fix changes the decoding of the BCD number, so that the version will also be `V6.0.0E`

To add a timestamp I have to think a little bit about a good solution...

onkelenno, 2024-07-10:
Sorry, I can't start the image, I'm on arm64.

s-allius, 2024-07-10:
Maybe it is possible for you to build an image from [issue119-branch](https://github.com/s-allius/tsun-gen3-proxy/tree/s-allius/issue119) for testting

onkelenno, 2024-07-11:
That fixed it, thank you.

s-allius, 2024-07-11:
fixed with #121

closed 2024-07-11 by s-allius

s-allius, 2024-07-28:
> 
> BTW: it would be cool if the topics tsun/# would also publish a timestamp so one could use persistant sessions.
> 
> Thank you

I just started to implement timestamp for MQTT. It is now planed for version 0.10.0.
For more details see: #138