s-allius/tsun-gen3-proxy #120: WLAN zum Inverter gen3 bricht ständig ab

Obi811, 2024-07-10:
Hallo,

ich habe seit version 0.8.0 ständige Abbrüche zum Inverter (5 - 10 Abbrüche pro Tag). Bei version 0.8.0 habe ich die cloud mal probehalber auf false gestellt und dann funktionierte es wieder für 3 Wochen. An der Konfiguration an der Fritz!Box und an Pihole habe ich nicht geändert und der lokale DNS passt auch. Im Moment läuft der proxy in der Version 0.10.0-preview.1 und dieser läuft auf einer Synology. Habe ich hier etwas übersehen? Ich hänge mal die Logdatei von der Synology an.
[tsun-proxy.csv](https://github.com/user-attachments/files/16164577/tsun-proxy.csv)
[trace.log](https://github.com/user-attachments/files/16164583/trace.log)
[proxy.log](https://github.com/user-attachments/files/16164594/proxy.log)
]

Was mich hier auch wundert, ist die Ip-Adresse 172.19.0.2 und 172.19.0.1, da ich hier eine feste IP mit 192.168.178.36 vergeben habe.
Und morgens bei der ersten Verbindung schickt der Proxy sporadisch den Wert vom Vortag komplett zu HA.
Mit freundlichen Grüßen
Andreas Obermayr

jscholtes, 2024-07-10:
Probleme mit dem WLAN habe ich seit dem  800W firmware Update auch.
Morgens läuft es für 1-2h, dann ist die WLAN Verbindung weg. Manchmal kommt sie dann irgendwann mal wieder

s-allius, 2024-07-10:
Die Verbindungsabbrüche sehe ich bei mir auch. Seit dem letztem Update baut der Wechselrichter etwa einmal pro Stunde die Verbindung ab und dann auch (meistens) wieder auf. Teilweise sendet der Wechselrichter keine Daten mehr, obwohl die TCP Verbindung zur Proxy steht. Auch bei mir ist es so, dass dies ich nicht ändert wenn ich die Verbindung vom Proxy zur Cloud abschalte.

Ich habe daraufhin in der Version 0.9 des Proxies eine Verbindungsüberwachung eingebaut, die erkennt wenn keine Daten mehr kommen und dann die Verbindung aktiv abbaut und so den Wechselrichter zu einem Verbindungsaufbau zwingt. Dies funktioniert aber auch nicht immer. Gefühlt denke ich es ist bei hoher Stromproduktion schlimmer. Evtl. ein EMV Problem, was die Verbindung stört und die Wechselrichter firmware nicht mehr behoben bekommt.

Mit der Proxie Version 0.9 ist es bei mir deutlich besser geworden, aber nicht fehlerfrei. Ich sehe keine Verbesserungsmöglichkeit mehr, ausser man könnte eine alte Firmware wieder einspielen. Ich habe aber nur das letze Update abspeichern können und daher kein älteres Images, welches diese Fehler evtl. nicht hat.

Wenn du schon die Proxie Version 0.9 laufen hast, kannst du das Problem nur bei TSUN reklamieren und auf eine bessere Firmware hoffen.

Obi811, 2024-07-10:
Vielen Dank für die schnelle Antwort. Ich habe jetzt nochmals den deutschen TSUN Support angeschrieben, und hoffe, das ich diesmal eine Antwort bekomme. Falls ich was bekomme, geb ich hier bescheid.

s-allius, 2024-07-10:
Hier noch ein Bild von meinem Setup. Da kann man sehen, dass regelmäßig der GEN3 Wechselrichter die Verbindung abbaut und ab und zu die Verbindung auch länger ausfällt.

[image]

Noch besser ist der Effekt zu sehen, wenn man sich den Graphen zu "Power on Time" anschaut. Dies sollte ein stetig steigender Graph sein. Man sieht aber das der Wechselrichter regelmäßig neu startet und manchmal tritt ein waagerecht Linie auf, dann liefert der Wechselrichter keine Daten mehr:
[image]

Obi811, 2024-07-10:
heute bin ich erst am Abend heimgekommen und da habe ich ihn verbunden, der erste Abbruch war schon heute früh um ca. 5:30 Uhr, nur hatte ich da keine Zeit mehr, ihn zu verbinden.
Ich habe jetzt noch nicht verfolgt, wie die App die WLAN Daten dann in den Inverter schreibt, aber gäbe es da eventuell eine Möglichkeit, von der Ferne aus sozusagen die WLAN Konfiguration vom Inverter anzustoßen und neu zu schreiben?

Hier mal meine Übersicht von heute
[image]

s-allius, 2024-07-11:
Mit der Version 0.8 ist ja die regelmäßige Abfrage der Inverter Register per Modbus reingekommen. Evtl. mag dies dein Wechselrichter nicht. Du hast auch eine andere Firmware verglichen mit meinem.
Ich werde mal einen Schalter einbauen, dass du dies deaktivieren kannst. Dann sehen wir ja, ob es da einen Zusammenhang gibt.

jscholtes, 2024-07-11:
> Gefühlt denke ich es ist bei hoher Stromproduktion schlimmer. Evtl. ein EMV Problem, was die Verbindung stört und die Wechselrichter firmware nicht mehr behoben bekommt.

Mein Wechselrichter hängt in der Garage und bis zum Accesspoint sind zwei Mauern dazwischen.
Ich habe jetzt gestern mal einen Repeater in die Garage gehängt und heute hatte ich den ganzen Tag noch keinen WLAN-Abbruch. 
Der MS600 wird unter Last auch sehr heiss. Vielleicht entlötet sich ja die WLAN-Antenne selber ;)

s-allius, 2024-07-11:
Ich habe eine Testversion gebaut, in der du die regelmäßigen Modbus Request abschalten kannst.

Dazu bitte das Image
```
    image: docker.io/sallius/tsun-gen3-proxy:dev
```
nutzen und in der `config.toml` eine Zeile mit `modbus_polling = false`in der Inverter Section hinzufügen:
```toml
[inverters."R17E7307021D006A"]
node_id = 'inv_1'               # MQTT replacement for inverters serial number  
suggested_area = 'Garage'  # suggested installation place for home-assistant
modbus_polling = false
``` 

Damit sollte sich der Proxy wieder wie vor der Version 0.8.0 verhalten.

Obi811, 2024-07-11:
Vielen Dank für die Version. Ich habe Sie soeben installiert und werde morgen berichten.

Obi811, 2024-07-12:
Also, der Inverter hat heute wieder einwandfrei verbunden und auch gehalten.
TSUN hat sich bei mir heute auch gemeldet und nach 2 Monaten um ca. 11 Uhr ein Update von V5.0.11 auf V5.1.09 und von RSW_400_V1.00.06 auf RSW_400_V1.00.17 durchgeführt. Soll ich hier noch was testen oder zukommen lassen?

s-allius, 2024-07-12:
Ich habe hier die gleichen Kombis an Versionen laufen, daher benötige ich keine Trace. Du kannst aber ausprobieren, ob es durch das deaktivieren des Modbus Pollings stabiler wird.

s-allius, 2024-07-14:
Das Abschalten des `MODBUS Polling`, zur besseren Untersuchung, kommt mit #126

closed 2024-07-14 by s-allius