s-allius/tsun-gen3-proxy #180: Run as standalone client app

il-steffen, 2024-09-02:
Thanks for your project!

My inverter powered up this morning and thanks to your project I was able to connect + setup wifi without (so far) having to go through TSUN/Orbit app registration stuff.

However I have HA and MQTT and many other things already running and I'm not a fan of "managing" applications with Docker. The new client mode also seems to make many things obsolete - I just need a client to poll the inverter and push the data to my favorite data base. One tool for one job.

I ran into a few issues trying to get this working and listed my detailed steps/problems below. Maybe this will be helpful for others or to improve the app. Also appreciate any comments if something can be done differently.

**1. Attempt to install basic python app instead of Docker stuff**
1. `python3 -m venv .venv; source .venv/bin/activate; pip -r requirements.txt `
 1. app entry point at `app/src/server.py`, example config in `app/config`
 1. I got a `Module not found tomllib` - we seem to require `>=python3.11`
 2. next we die on logging: `logging.config.fileConfig('logging.ini')` - need to run from within app/src
 3. failed to open `log/proxy.log` - need to create `log/`
 4. failed to open `default_config.toml`, after copying this from `../config` it tries to `shutil.copy2()` it to config/, then again fails to open and advises about renaming some example.toml. So actually we need to copy to `app/src/config/config.toml`...

**2. Configuration file...I think this could use some background to explain the purpose of each section**
1. reach tsun/solarman cloud....why do we care? just for FW updates?
3. home assistant config could use 1-2 sentences background info as well. The comments sound like you're actually listening/checking on some HA status updates?
4. `inverter mapping` looks reasonable, except I guess its unused in 'client_mode'. I could see my TSUN MS2000 pop up in HA MQTT while the two inverters in the default config don't seem to exist
6. what does modbus_polling do? is this just for specific inverters, do we need special cabling, does it give us extra information or alternative way to access the device instead of wifi/proxy?
7. client_mode...here we go...it just works, great! I can see TSUN/MS2000 info and my panels. You seem to grab all the meta-data from the server. Is there even anything to configure in this case? I don't even see the string 'micro inverter' in the config. Is this 'client_mode' config part of the `inverts.Y17...` clause above or can I delete them? Would it support multiple `client_mode` configs and list them out with different names?
8. `pv=`...are these configs going to be mapped to the 4 panels found on my MS2000?
9. access control...these are super obscure and only documentation is the example config itself. I guess these are actually setting filters on AT commands seen by your proxy? So...it doesn't apply in client-mode, and I guess its also not relevant if tsun/solarman is not `enabled`?

Need to play around some more...to be continued..

s-allius, 2024-09-03:
Hi il-steffen,

thank you for your great feedback. I will have a deep look at it at the weekend and will use it for improvements...

Cheers Stefan

s-allius, 2024-09-07:
Hi il-steffen,

at the start of the project I had decided to publish this project only as a Docker image. This way I avoid dependency issues as the container contains everything that is needed. For development, I always use the latest Phython release (currently 3.12.5), which also runs all unit tests. I don't bother with older versions, as I don't think this is necessary for containerization.

## Configuration
2.1 You are right, you only need this for updates or if you want to use the TSUN apps. From a security point of view, it would be better not to configure these connections, as this sends a lot of data to TSUN (e.g. the network configuration, ...) 

2.2 Yes, I listen for the Home Assistant status in these MQTT topics. When the status changes to online, I register all devices with Home Assistant. With this approach, I don't need to send topics as retaining messages. This avoids a lot of problems

2.3 At the moment, you only need to configure the IP address and the port in 'Client Mode' (if it is not the default value 8899). The rest is for later proxy versions, or when not using `Client Mode`. For example, we could implement a connection to the TSUN Cloud even if the inverter is running in `Client Mode`. Then we can filter critical information like `WiFi password`, which is not possible with an SSL connection to TSUN.

2.4 `Modbus polling`is for inverters which are connected in `server mode` to get information faster from the device. In server mode the inverter sends data only every 5 minutes. With the polling you get data every 60 seconds. 
The client mode use the `Modbus polling` automatically, since the inverter doesn't send any data in this mode. The data polling time is about 30 seconds then. See also: https://github.com/s-allius/tsun-gen3-proxy/wiki/Operation-Modes-Overview

il-steffen, 2024-09-07:
Too sunny for work today but if you're open to it, I would take a shot at offering a pip install option as well. Should also work fine with CI.

Intel Deutschland GmbH
Registered Address: Am Campeon 10, 85579 Neubiberg, Germany
Tel: +49 89 99 8853-0, www.intel.de
Managing Directors: Sean Fennelly, Jeffrey Schneiderman, Tiffany Doon Silva
Chairperson of the Supervisory Board: Nicole Lau
Registered Office: Munich
Commercial Register: Amtsgericht Muenchen HRB 186928

s-allius, 2024-09-07:
Yes, with pleasure. I just made it easy for myself, as Phython and Docker were new territory for me.  I didn't bother with modules and the Pip installation.
A pip installation is certainly a welcome alternative for many, as Docker requires a lot of knowledge.

If you need help or have any questions, I'm here to help

s-allius, 2024-12-23:
I fixed the file handling, so the server can easily started from any directory. If the mandatory log directory is missing, the proxy creates this.

implemented with #265

closed 2024-12-23 by s-allius