s-allius/tsun-gen3-proxy #206: tsun proxy as home-assistant add-on

mime24, 2024-11-03:
Thank you for this inspiring work! 
since some weeks I'm curious how this project develops and whether there is an alternative way to install. For users like me who are running a supervised HA instance a home assistant add-on would be grat. it seems super close to the development of containers but are easier to use for the majority of HA users. I'm not a developer, so I have no idea about the milestones to achieve for running the proxy as an add-on. 
It would be just the proxy as HA is obviously already there. the broker is also not needed as there is an existing add-on.

have you thought about this option? do you think this is achievable?

s-allius, 2024-11-04:
Hello mime24,

the aim of this project is to create a solution that is as universal as possible. This means that the proxy only takes care of the data processing and provides it via a standard protocol such as MQTT. This enables the proxy to be used for various automation systems. Home Assistant is just one possibility.

It may well be that it is not that difficult to integrate the proxy into HA. But I'm not familiar with the internals of HA. For me it is important that dependencies to other software parts are well isolated. An intermediate layer would therefore have to be created that adapts the Ethernet and MQTT interface. I can't say whether this makes sense. If someone would like to take on this, I would be happy to help. But I don't have the resources myself.
Setting up a Docker is not that difficult either...

mime24, 2024-11-05:
understood your motivation/concerns. let me check whether I can observe the required steps, try something and maybe also find someone else who could support.

you're right that setting up docker is not so difficult. But for most of the home assistant users (>77% running on Home Assistant OS) the docker cannot/shouldn't executed on the host where HA is running. You would not be able to update your HA installation anymore due to compliance issues. 
_"If you run Home Assistant, please be aware that running additional containers is not a supported use-case and will trigger your system to be flagged as unsupported as well."_

mime24, 2024-11-10:
Hi @s-allius, yesterday I did some research. I quickly discovered that add-ons are also created using Dockerfiles. It took me a while to understand how it all works. It quickly became clear to me that, in theory, no major changes were needed to get it up and running. In practice, I was stumped for a very long time. I just hadn't managed to install the necessary libraries via pip. apparently, the libraries either have to be available in the host system or you have to use a virtual python environment. In the end it worked.
The server is running and shoveling diligently for Homeassistant.
Due to my lack of knowledge, I had to improvise a bit. The solution is therefore not yet particularly attractive. 

ToDo's would still be among others:

- Config via add-on config tab
- use MQTT user info via HA-variable 
- check file structure and harmonize
- check Dockerfile
- check how to load the files when installing from your repo instead of keeping a copy

s-allius, 2024-11-10:
Hey, that's sounds good.

To harmonize some file structure should be a problem for me. Maybe we can reach in the MQTT user setup via an environment setting. That would be the typical way for containers. 

If you have a description, or more details which values we have to pass, I can add it.

mime24, 2024-11-13:
I'm currently trying to clean up the add-on, create some info and comments to the code. How shall I handle the add-on on Github. I'm not familiar with all that stuff, so I'm asking you whether I shall create an own Projekt or is it better to add it somehow to yours?

s-allius, 2024-11-27:
Ich habe das AddOn im Container unter vscode einmal starten können und habe das gleiche Problem wie du gesehen.

Allerdings ist bei mir die Umgebung nicht stabil, oder ich habe den Umgang noch nicht richtig verstanden. Nach dem Versuch ein Rollback der Sourcen bekomme ich HA gar nicht mehr gestartet. Mir ist noch nicht klar, wie überhaupt das Szenario gebaut wird. Kann auch an den Ressourcen meines Laptops liegen (MacBook mit Intel-I5 8GB). War bisher kein Problem, da ich ja auch nur die Container zusammengebaut habe. 
Selbst wenn ich die ältere Sourcen auf der Platte habe, findet er nur da Plugin mit der letzen Versionsnummer. Scheint also etwas irgendwo gecacht zu werden.
Da muss ich noch etwas Zeit investieren. Heute bin ich aber von der Familie gebucht :-)

closed 2024-12-03 by s-allius