s-allius/tsun-gen3-proxy #215: HA AddOn: Concept how to handle the proxy logs in an AddOn

s-allius, 2024-12-02:
The proxy writes a lot of information in log files. These are huge (5..8MB per day). We need a concept how we would handle this:

- we can disable all log
- we can automatically delete the files after x days
- we can use ...?

mime24, 2024-12-07:
I mounted the add-on log folder for testpurpose. but, this is something we could take over to the actual add-on config. this would give users the possibility to see the logs.
this in combination with a switch to enable/disable might be  meaningful.

let me keep this test for some days and I'll come with a more precise suggestion.

s-allius, 2024-12-17:
- I created a separate issue for this: https://github.com/s-allius/tsun-gen3-proxy/issues/250
- I also disable the watchdog for now

closed with #243

closed 2024-12-17 by s-allius