s-allius/tsun-gen3-proxy #219: HA AddOn: Complete. the configuration

s-allius, 2024-12-02:
There is a new emulator mode comming with version 0.12

We have to extend the config for this.

mime24, 2024-12-04:
let me know what's missing or where to find the parameters to be added.

s-allius, 2024-12-06:
Hi Michael,
I'm preparing a new config reader class in the proxy. The new one will be able to read the options.json directly and also supports a converter which resolve the dot-notation in the keywords. This still works and I have verified this with unit tests. 
Now I just need to add an environment reader so that we can also read the MQTT variables directly. That will be ready soon, and then we can continue with these tickets.

s-allius, 2024-12-18:
There is still the configuration of the client-mode missing. The client mode is useful if the inverter only supports encrypted connections or if there are other reasons why the connection from an inverter can't be redirected to the proxy. The `forward` flags enabled an inverter simulation, which is used for connecting the proxy at the TSUN or Solarman Cloud. Then the proxy can forward the values which he get from the inverter to the cloud.  

to configure the client mode you can easily add something like:
```toml
client_mode = {host = '192.168.80.49', port = 8899, forward = true}
```
to the inverter block.

s-allius, 2024-12-22:
implemented with #252

closed 2024-12-22 by s-allius