s-allius/tsun-gen3-proxy #234: Daily Generation Wert zu hoch

Obi811, 2024-12-08:
Hallo, ich hatte vor ein paar Wochen mehrmalige kurze Stromausfälle. Auf einmal wurde der Wert Daily Generation von beiden Solarplatten zu hoch.
Ich habe es erst auf HA geschoben und gestern neu aufgesetzt. Der Proxy läuft in der Version 0.11.1.
Ist hier etwas bekannt oder gibt es hier Abhilfe?

Mit freundlichen Grüßen
Andreas Obermayr
[image]
[image]
[image]

s-allius, 2024-12-08:
Hallo Andreas,

nein mir ist da bisher nichts bekannt.
Du kannst mal in die Log Files schauen. im Log Verzeichnis gibt es ein File `proxy.log`. Darin mal nach `total.Daily_Generation`suchen und schauen, ob es da nicht plausible Werte gibt.

Die Einträge müsste so:
```
'2024-12-08 10:43:55  INFO | data | [inv_1/] GEN3: total.Daily_Generation : 0.01kWh'
```
oder so:
```
'2024-12-08 10:46:17  INFO | data | [inv_2/] MODBUS: total.Total_Generation : 799.86kWh'
```
ausschauen. Je nachdem aus welcher Quelle sie stammen.

Was für einen Inverter und Fw hast du?

Obi811, 2024-12-08:
Hallo Stefan,
Vielen Dank für die schnelle Antwort.
ich finde die Werte die hier stehen, sind viel zu hoch.
Habe Sie mal mit angehängt.

[proxy.log](https://github.com/user-attachments/files/18051731/proxy.log)

Als Inverter habe ich den TSO-MS800
Firmware ist RSW_400_V1.00.17
und V5.1.09

s-allius, 2024-12-08:
Interessant, hier ist ein Sprung:

```
'2024-12-08 07:55:09  INFO | data | [inv1/] GEN3: total.Daily_Generation : 0.62kWh'
'2024-12-08 07:55:10  INFO | data | [inv1/] GEN3: total.Daily_Generation : 2.01kWh'
```
Es wäre gut mal ein log des Traffic zu machen wenn so ein Sprung auftritt. Dazu kann ich dir einen Debug-Container bauen. Dann sehen wir auch die Pakete die vom Inverter kommen und können den Fehler besser einkreisen. Dauert aber ein paar Minuten, da mein Laptop nicht der schnellste ist...

s-allius, 2024-12-08:
So der Container ist fertig :-) 
Du kannst ihn von GitHub laden, indem du in deiner composer.yaml folgendes Image angibst:
```yaml
    image: docker.io/sallius/tsun-gen3-proxy:debug
```
Dann kannst du in der `proxy.log` Datei nach einem Spring des Wertes schauen und mit dem Zeitstempel im `trace.log` File die Stelle suchen. Wir benötigen einen Zeitausschnitt von etwa 11-12Minuten um den Sprung herum

Obi811, 2024-12-08:
Habe es gerade aktualisiert.
Ich beobachte jetzt mal und melde mich wieder.
Vielen Dank schon mal

s-allius, 2024-12-08:
Für den Trace-log muss man das properitäre Protokoll des Inverters kennen. Ich habe per revers Engineering, das meister verstanden. Scheue mich aber das öffentlich zu dokumentieren...

Ich schaue mir deine Logs an. Werde ich aber vermutlich nicht mehr heute schaffen. Ich melde mich wenn ich mehr weiß

s-allius, 2024-12-10:
Ich habe mir die Logs genau angeschaut Um 12.41:19 gab es einen Sprung in den Werten. Die Dekodierung Funktion also richtig. Was aber auffällt ist, dass in dem Datenpaket am Anfang ein paar Register fehlen und dafür am Ende ein paar mir unbekannte Byte anhängen.
Es könnte also sein, dass der Inverter hier alte Werte wiederholt. Ist aber nur eine Vermutung. 

Siehst du die Sprünge in der TSUN App auch?

Obi811, 2024-12-10:
Die TSUN App funktioniert und zeigt die richtigen Werte an, hier sind keine Sprünge drin

s-allius, 2024-12-10:
In den Pakete ist auch noch ein Zeitstempel kodiert In dem Ersten ist das aktuelle Datum und in dem Zweiten ein 5 Monate alter:
```
00000193a649d9b0
GMT: Sunday, 8. December 2024 12:41:18
Your time zone: Sonntag, 8. Dezember 2024 13:41:18 GMT+01:00
--

000001909855e748
GMT: Tuesday, 9. July 2024 16:31:25
Your time zone: Dienstag, 9. Juli 2024 18:31:25 GMT+02:00 DST
```

Obi811, 2024-12-10:
finde ich komisch, ich weiß, das TSUN am 12. Juli ein Update auf diese Firmware gemacht hat, um die Zeit hatte ich auch Verbindungsabbrüche, siehe Issue #120

s-allius, 2024-12-16:
fixed with #246