s-allius/tsun-gen3-proxy #238: HA AddOn: Repository check

s-allius, 2024-12-08:
I created a repository for the add-ons inside this project. The mandatory files are in this folder:

```bash
https://github.com/s-allius/tsun-gen3-proxy/tree/main/ha_addons
```
And there is a add-on which loads the following image:
```bash
docker.io/sallius/tsun-gen3-addon:dev
```

- A test must be made if the approach is functional and if we can remove the `Dockerfile`

s-allius, 2024-12-09:
Ahh, da wit ein git clone durchgeführt. Probier doch mal folgenden Eintrag

```
https://github.com/s-allius/tsun-gen3-proxy
```
Wenn das auch nicht funktioniert muss ich vermutlich ein eigenes Repro dafür anlegen..-

Obi811, 2024-12-09:
Klappt auch nicht
[image]

s-allius, 2024-12-10:
Ja, du hast recht das yaml file war nicht im root. Habe es verschoben. Bin gespannt, ob das AddOn jetzt gefunden wird. wenn nicht mache ich ein extra Projekt, damit die Verzeichnisstruktur nicht völlig kaput geht.

Obi811, 2024-12-10:
Addon wird über beide links gefunden, habe jetzt den letzen mal dringelassen. Addon lässt sich auch installieren.
habe es nun konfiguriert.
Oben beim Änderungsprotokoll wenn ich draufklicke kommt die Meldung No changelog found for add-on c676133d_tsun-proxy!

Hier habe ich mal zusätzlich den Log mit angehängt inkl. einem Neustart des Addons
`
 Home Assistant Supervisor: 2024.11.4
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Add-on environment started
check for Home Assistant MQTT
MQTT found
Start Proxyserver...
'2024-12-10 20:30:54  INFO | root | Server "proxy - 0.12.0-dev" will be started'
'2024-12-10 20:30:54  INFO | root | AddOn: False'
'2024-12-10 20:30:54  INFO | root | config_path: ./config/'
'2024-12-10 20:30:54  INFO | root | json_config: /data/options.json'
'2024-12-10 20:30:54  INFO | root | toml_config: None'
'2024-12-10 20:30:54  INFO | root | LOG_LVL    : INFO'
'2024-12-10 20:30:54  INFO | root | ******'
'2024-12-10 20:30:54  INFO | root | Read from default_config.toml => ok'
'2024-12-10 20:30:54  INFO | root | Read from Read environment => ok'
'2024-12-10 20:30:54  INFO | root | Read from ./config/config.json => n/a'
'2024-12-10 20:30:54  INFO | root | Read from ./config/config.toml => n/a'
'2024-12-10 20:30:54  INFO | root | Read from /data/options.json => ok'
'2024-12-10 20:30:54  INFO | root | ******'
'2024-12-10 20:30:54  INFO | mqtt | start MQTT: host:core-mosquitto  port:1883  user:addons'
'2024-12-10 20:30:54  INFO | root | HTTP server listen on port: 8127'
'2024-12-10 20:30:54  INFO | mqtt | MQTT broker connection established'
'2024-12-10 20:30:54  INFO | root | Initialize proxy device on home assistant'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
'2024-12-10 20:31:37  INFO | root | Shutdown due to SIGTERM'
'2024-12-10 20:31:37  INFO | root | Proxy disconnecting done'
'2024-12-10 20:31:37  INFO | root | HTTP server cancelled'
'2024-12-10 20:31:37  INFO | root | Event loop is stopped'
'2024-12-10 20:31:37  INFO | root | Close MQTT Task'
'2024-12-10 20:31:37  INFO | root | Finally, exit Server "proxy"'
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: TSUN-Proxy
 MQTT Proxy for TSUN Photovoltaic Inverters
-----------------------------------------------------------
 Add-on version: dev
 You are running the latest version of this add-on.
 System: Home Assistant OS 14.0  (amd64 / qemux86-64)
 Home Assistant Core: 2024.12.1
 Home Assistant Supervisor: 2024.11.4
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Add-on environment started
check for Home Assistant MQTT
MQTT found
Start Proxyserver...
'2024-12-10 20:32:01  INFO | root | Server "proxy - 0.12.0-dev" will be started'
'2024-12-10 20:32:01  INFO | root | AddOn: False'
'2024-12-10 20:32:01  INFO | root | config_path: ./config/'
'2024-12-10 20:32:01  INFO | root | json_config: /data/options.json'
'2024-12-10 20:32:01  INFO | root | toml_config: None'
'2024-12-10 20:32:01  INFO | root | LOG_LVL    : INFO'
'2024-12-10 20:32:01  INFO | root | ******'
'2024-12-10 20:32:01  INFO | root | Read from default_config.toml => ok'
'2024-12-10 20:32:01  INFO | root | Read from Read environment => ok'
'2024-12-10 20:32:01  INFO | root | Read from ./config/config.json => n/a'
'2024-12-10 20:32:01  INFO | root | Read from ./config/config.toml => n/a'
'2024-12-10 20:32:01  INFO | root | Read from /data/options.json => ok'
'2024-12-10 20:32:01  INFO | root | ******'
'2024-12-10 20:32:01  INFO | mqtt | start MQTT: host:core-mosquitto  port:1883  user:addons'
'2024-12-10 20:32:01  INFO | root | HTTP server listen on port: 8127'
'2024-12-10 20:32:01  INFO | mqtt | MQTT broker connection established'
'2024-12-10 20:32:01  INFO | root | Initialize proxy device on home assistant'`

s-allius, 2024-12-10:
OK, das ist ja schonmal gut. Ich bin mir noch nicht sicher, ob es besser ist das AddOn-Repository als separates Projekt in Github zu haben, oder ob wir es doch Zusammenlassen?
Im Grunde finde ich es charmant, wenn alles zusammen bleibt. Ist vermutlich einfacher zu pflegen

Für die Logs, könnte das Verzeichnis auch konfigurierbar machen.

s-allius, 2024-12-11:
Ich habe jetzt noch ein zweites Add-On in das zweite Repro gelegt. Es handelt sich um die Debug-Version mit eingeschaltendem DEBUG Log-Level. Deine DOC File habe ich auch rein kopiert.

Wenn das so funktioniert, kümmere ich mich als nächstes um das Konfigurieren des Log-Paths.

Ach ja, die beiden Container sind noch mit den alten Dockerfiles gebaut, da ich die noch nicht gemerged habe.

Obi811, 2024-12-11:
Ein zweites Addon wird mir nicht angezeigt, es war nur eine Aktualisierung, statt dev steht jetzt debug.
habe es aktualisiert, hier das neue Log:

```
-----------------------------------------------------------
 Add-on: TSUN-Proxy (Debug)
 MQTT Proxy for TSUN Photovoltaic Inverters with Debug Logging
-----------------------------------------------------------
 Add-on version: debug
 You are running the latest version of this add-on.
 System: Home Assistant OS 14.0  (amd64 / qemux86-64)
 Home Assistant Core: 2024.12.1
 Home Assistant Supervisor: 2024.11.4
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Add-on environment started
check for Home Assistant MQTT
MQTT found
Start Proxyserver...
'2024-12-11 05:33:05  INFO | root | Server "proxy - 0.12.0-debug" will be started'
'2024-12-11 05:33:05  INFO | root | AddOn: False'
'2024-12-11 05:33:05  INFO | root | config_path: ./config/'
'2024-12-11 05:33:05  INFO | root | json_config: /data/options.json'
'2024-12-11 05:33:05  INFO | root | toml_config: None'
'2024-12-11 05:33:05  INFO | root | LOG_LVL    : INFO'
'2024-12-11 05:33:05  INFO | root | ******'
'2024-12-11 05:33:05  INFO | root | Read from default_config.toml => ok'
'2024-12-11 05:33:05  INFO | root | Read from Read environment => ok'
'2024-12-11 05:33:05  INFO | root | Read from ./config/config.json => n/a'
'2024-12-11 05:33:05  INFO | root | Read from ./config/config.toml => n/a'
'2024-12-11 05:33:05  INFO | root | Read from /data/options.json => ok'
'2024-12-11 05:33:05  INFO | root | ******'
'2024-12-11 05:33:05  INFO | mqtt | start MQTT: host:core-mosquitto  port:1883  user:addons'
'2024-12-11 05:33:05  INFO | root | HTTP server listen on port: 8127'
'2024-12-11 05:33:05  INFO | mqtt | MQTT broker connection established'
'2024-12-11 05:33:05  INFO | root | Initialize proxy device on home assistant'
'2024-12-11 07:55:49  INFO | conn | [G3:0] Accept connection from ('192.168.178.36', 62631)'
'2024-12-11 07:56:01  INFO | conn | [G3:1] Accept connection from ('192.168.178.36', 51225)'
'2024-12-11 07:56:14  INFO | conn | [G3:2] Accept connection from ('192.168.178.36', 59007)'
'2024-12-11 07:56:27  INFO | conn | [G3:3] Accept connection from ('192.168.178.36', 60324)'
'2024-12-11 07:56:40  INFO | conn | [G3:4] Accept connection from ('192.168.178.36', 52380)'
'2024-12-11 07:56:52  INFO | conn | [G3:5] Accept connection from ('192.168.178.36', 59373)'
'2024-12-11 07:57:05  INFO | conn | [G3:6] Accept connection from ('192.168.178.36', 54916)'
'2024-12-11 07:57:18  INFO | conn | [G3:7] Accept connection from ('192.168.178.36', 64389)'
'2024-12-11 07:57:31  INFO | conn | [G3:8] Accept connection from ('192.168.178.36', 55936)'
'2024-12-11 07:57:43  INFO | conn | [G3:9] Accept connection from ('192.168.178.36', 57096)'
'2024-12-11 07:57:55  INFO | conn | [G3:10] Accept connection from ('192.168.178.36', 63351)'
'2024-12-11 07:58:08  INFO | conn | [G3:11] Accept connection from ('192.168.178.36', 54737)'
'2024-12-11 07:58:20  INFO | conn | [G3:12] Accept connection from ('192.168.178.36', 56093)'
'2024-12-11 07:58:32  INFO | conn | [G3:13] Accept connection from ('192.168.178.36', 56726)'
'2024-12-11 07:58:46  INFO | conn | [G3:14] Accept connection from ('192.168.178.36', 51700)'
'2024-12-11 07:58:58  INFO | conn | [G3:15] Accept connection from ('192.168.178.36', 61263)'
'2024-12-11 07:59:10  INFO | conn | [G3:16] Accept connection from ('192.168.178.36', 51331)'
'2024-12-11 07:59:23  INFO | conn | [G3:17] Accept connection from ('192.168.178.36', 63237)'
'2024-12-11 07:59:36  INFO | conn | [G3:18] Accept connection from ('192.168.178.36', 49467)'
'2024-12-11 07:59:48  INFO | conn | [G3:19] Accept connection from ('192.168.178.36', 57286)'
'2024-12-11 08:00:01  INFO | conn | [G3:20] Accept connection from ('192.168.178.36', 53462)'
'2024-12-11 08:00:14  INFO | conn | [G3:21] Accept connection from ('192.168.178.36', 65335)'
'2024-12-11 08:00:27  INFO | conn | [G3:22] Accept connection from ('192.168.178.36', 51669)'
[... 348 lines cut ...]
'2024-12-11 09:59:51 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 09:59:52 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 09:59:53 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 09:59:54 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 10:05:10 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 10:05:11 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 10:05:12 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 10:05:13 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
[... cut ...]
'2024-12-11 10:10:38 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 10:10:39 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 10:10:40 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 10:10:41 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 10:10:42 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
'2024-12-11 10:10:43 ERROR | root | Infos.parse: data_type: 8 @0x000a No:157 not supported'
```

s-allius, 2024-12-14:
Hallo Michael,
Das Log sieht doch schonmal gut aus.
- die Errors sind die gleichen wie in #234. Da muss ich mir noch eine Lösung überlegen
- die Warnings sind Verbindungsabbrüche wie sie bei Dämmerung immer auftreten. Diese sollten Bei genug PV Erzeugung aber nicht da sein.
- für die Log-file configuration habe ich in #215 einen ersten Vorschlag.

Im Add-On Repro habe ich nun für die beiden Versionen zwei verschiedene Slugs benutzt. Siehst du jetzt zwei Versionen (dev und debug)?

Obi811, 2024-12-14:
Es sind jetzt zwei Versionen in der repro https://github.com/s-allius/ha_addons vorhanden
[image]

s-allius, 2024-12-14:
Das ist doch prima. Dann werde ich mal schauen, dass ich die verschiedenen Files aus Template-Files erzeuge, damit das ganze auch wattbar bleibt.

s-allius, 2024-12-15:
closed with #244

closed 2024-12-15 by s-allius