s-allius/tsun-gen3-proxy #255: Initialize proxy device on home assistant issue

lukaszlk, 2024-12-22:
Hi,

I’ve set up an addon but in logs it estsbilishes an MTQQ connection succesfully but then stops on „Initialize proxy device on home assistant„. Does anybody know how to resolve this issue? Thanks

Hagenncl, 2024-12-22:
Did you loop your DNS server? 
For combination of Home Assistant Core / Supervised you can use following guideline: 

**EN:** 

Here is a concise step-by-step guide to setting up **AdGuard Home** on **Home Assistant** and configuring the **FRITZ!Box 7490** to use AdGuard as the DNS server, including the use of the URL `logger.talent-monitoring.com`:

---

### **1. Install AdGuard Home on Home Assistant**
1. **Install AdGuard Home**:
   - Open **Supervisor** in Home Assistant.
   - Go to the **Add-on Store** and install **AdGuard Home**.
2. **Start AdGuard Home**:
   - Start the add-on and open its web interface (`http://<Home Assistant IP>:3000`).
3. **Configure AdGuard**:
     - **DNS Rewrites**:
     - Navigate to **Settings** > **DNS settings** > **DNS rewrites**.
     - Add the following rewrite:
       - **Domain name**: `logger.talent-monitoring.com`
       - **IP address**: The local IP address of your Home Assistant, e.g., `192.168.178.50`.

---

### **2. Find the AdGuard IP Address**
- Determine the local IP address of the device running Home Assistant (e.g., `192.168.178.50`).
- This IP will be used as the DNS server in the FRITZ!Box configuration.

---

### **3. Configure the FRITZ!Box to Use AdGuard as DNS Server**
1. **Access the FRITZ!Box**:
   - Open `http://fritz.box` in a browser and log in.
2. **Change DNS Server Settings**:
   - Navigate to **Internet** > **Access Data** > **DNS Server**.
   - Select **Use other DNSv4 servers**.
   - Enter the AdGuard IP address:
     - **Primary DNS server**: `192.168.178.50` (AdGuard IP).
     - **Secondary DNS server**: Leave blank or enter the same IP.
   - Save the changes.
3. **IPv6 DNS (Optional)**:
   - If you use IPv6, under **Use other DNSv6 servers**, enter the IPv6 address of AdGuard.

---

### **4. Configure DNS Rebind Protection in the FRITZ!Box**
1. Go to **Home Network** > **Network** > **Network Settings** tab.
2. Scroll down to the **DNS Rebind Protection** section.
3. Add an exception for the domain:
   ```
   logger.talent-monitoring.com
   ```
4. Save the changes.

---

### **5. Test the Configuration**
1. **Test DNS Resolution**:
   - On a device in the network:
     - Open a terminal or command prompt and run:
       ```
       nslookup logger.talent-monitoring.com
       ```
     - The response should return the IP address of Home Assistant (e.g., `192.168.178.50`).
2. **Check AdGuard Logs**:
   - Open AdGuard Home and go to **Query Log**.
   - Verify that queries to `logger.talent-monitoring.com` appear and are resolved correctly.

In addition you need to change the config of your HomeAssistant Network setting to root DNS server to external DNS Server. Therfore use this guide: 

**Home Assistant OS (Supervisor) - Configure DNS Servers**
Steps:
1.	Open the Web Interface:
        o	Access the Home Assistant interface via your browser (e.g., http://homeassistant.local:8123).
2.	Go to Settings:
        o	Navigate to Settings > System > Network.
3.	Select the Network Interface:
        o	Click on the active network interface (e.g., eth0 for wired connections or wlan0 for Wi-Fi).
4.	Configure DNS Servers:
        o	Scroll down to the DNS server section.
        o	First, delete the IP address pointing to the FRITZ!Box (e.g., 192.168.178.1), as it would route DNS traffic through the FRITZ!Box.
        o	Then, from the dropdown menu, select the option Google (8.8.8.8, 8.8.4.4).
5.	Save and Restart:
        o	Save the changes.
        o	Restart Home Assistant to apply the new DNS settings.
6.	Verification:
        o	Verify that the DNS settings are working by ensuring Home Assistant can access external websites or services like updates.

lukaszlk, 2024-12-22:
Hi,
Yes, the nslookup anawers as below
[image]
[image]

Maybe the issue could be connected to my Gen3Plus inverter MS-1800- in its settings i cannot change the adress and port of a cloud Access as the primary option is set on default with no option to change.
[image]
I can only add a B option with an IP of my homeassistant device and a 10000 port. Maybe the issue lays there?

closed 2024-12-23 by lukaszlk