s-allius/tsun-gen3-proxy #267: TSUN Proxy HA Addon und Adguard Addon keine Weiterleitung von Proxy zum offiziellen TSun Server

Obi811, 2024-12-26:
Hallo,
habe seit ein paar Tagen das Addon laufen und vor kurzem auf das Adguard-Addon umgestellt.
seit diesem zeitraum funktioniert die ofizielle TSUN App nicht mehr.
Habe es so eingerichtet wie in dem einen Artikel für Adguard.
In den Logs kommt dann diese Meldung
```
'2024-12-26 12:02:51 ERROR | root | resolve logger.talent-monitoring.com to 192.168.178.22, which is a private IP!
  Check your DNS settings and use a public DNS resolver!
  To prevent a possible loop, forwarding to local IP addresses is
  not supported and is deactivated for subsequent connections
'
``` 

Was mir aufgefallen ist, das HA auch dann über Adguard läuft und dieser will wahrscheinlich meiner Vermutung auch wieder auf den Proxy umschlüsseln.

Ich habe jetzt gerade in Home assistant unter Einstellungen / System / Netzwerk / IPV4 und IPV6 statt der Adguard DNS, welche automatisch eingefügt wurde einen öffentlichen eingefügt und hoffe hier Besserung.

Sobald morgen wieder die Sonne da ist, berichte ich, ob diese Einstellung reicht oder würdet ihr das anders lösen?

Verbindung von Inverter zum Proxy funktioniert einwandfrei.

Mit freundlichen Grüßen
Andreas Obermayr

mime24, 2024-12-26:
Hallo Andreas, genau daran sollte es liegen. Der Homeassistant sollte immer eine feste IP Konfiguration und externen DNS Server haben. (Darauf wird in der Installationsanweisung von AdGuard Home Addon auch nochmal explizit hingewiesen) 
Ich gehe somit davon aus, dass morgen alles wie gewünscht funktioniert.

s-allius, 2024-12-26:
Hallo Andreas,
du musst nachdem du den Resolver umgestellt hast auch einmal deinen TSUN Proxy neustarten, damit er die Configuration neu liest und die Weiterleitung wieder aktiviert.

Obi811, 2025-01-07:
Hallo, sorry für die späte Meldung...
nach dieser Umstellung klappt alles, nur das ihr bescheid wisst.

closed 2025-01-07 by s-allius