s-allius/tsun-gen3-proxy #269: HA Addon: Multiple inverters

frankol, 2025-01-11:
Ich versuche mehrere Wechselrichter in Home Assistant einzubinden, jedoch kommt da irgendwie Grütze raus.

Konfiguration im Addon sieht so aus (Seriennummern geändert):

```
- serial: Y47E00000020000CE
  monitor_sn: '2090000186'
  node_id: PV-Garage-links
  suggested_area: Garage
  modbus_polling: true
  client_mode.host: 192.168.200.49
  client_mode.port: 8899
  client_mode.forward: false
  pv1.manufacturer: Shinefar
  pv1.type: SF-M18/144550
  pv2.manufacturer: Shinefar
  pv2.type: SF-M18/144550

- serial: Y47E00000200000B
  monitor_sn: "2090000183"
  node_id: PV-Garage-mitte
  suggested_area: Garage
  modbus_polling: false
  client_mode.host: 192.168.200.50
  client_mode.port: 8899
  client_mode.forward: false
  pv1.manufacturer: Shinefar
  pv1.type: SF-M18/144550
  pv2.manufacturer: Shinefar
  pv2.type: SF-M18/144550
```
Am MQTT kommt aber sowas an:
[image]

Der erste Wechselrichter wird unter PV-Garage-mitte einsortiert und der zweite ist einfach unter tsun.
Damit stimmen auch die Werte nicht von den einzelnen Geräten. Es werden z.B. an allen Wechselrichtern die gleiche Leistung angezeigt.

Ich habe versucht die config.toml unter /homeassistant/tsun-proxy/ anzulegen, aber diese wird nicht eingelesen:

```
'2025-01-11 16:02:22  INFO | root | Read from /homeassistant/tsun-proxy/config.json => n/a'
'2025-01-11 16:02:22  INFO | root | Read from /homeassistant/tsun-proxy/config.toml => n/a'
```

Es sind übrigens TSOL-MS800 Wechselrichter mit einem Webinterface, sprich GEN3Plus.

Hier noch ein Auszug aus dem Log:

```
-----------------------------------------------------------
 Add-on: TSUN-Proxy
 MQTT Proxy for TSUN Photovoltaic Inverters
-----------------------------------------------------------
 Add-on version: 0.12.0-rel
 You are running the latest version of this add-on.
 System: Home Assistant OS 14.1  (amd64 / generic-x86-64)
 Home Assistant Core: 2025.1.1
 Home Assistant Supervisor: 2024.12.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Add-on environment started
check for Home Assistant MQTT
MQTT found
Start Proxyserver...
'2025-01-11 12:21:53  INFO | root | Server "proxy - 0.12.0-rel" will be started'
'2025-01-11 12:21:53  INFO | root | current dir: /home/proxy'
'2025-01-11 12:21:53  INFO | root | config_path: /homeassistant/tsun-proxy/'
'2025-01-11 12:21:53  INFO | root | json_config: /data/options.json'
'2025-01-11 12:21:53  INFO | root | toml_config: None'
'2025-01-11 12:21:53  INFO | root | log_path:    /homeassistant/tsun-proxy/logs/'
'2025-01-11 12:21:53  INFO | root | log_backups: 2 days'
'2025-01-11 12:21:53  INFO | root | LOG_LVL    : INFO'
'2025-01-11 12:21:53  INFO | root | ******'
'2025-01-11 12:21:53  INFO | root | Read from default_config.toml => ok'
'2025-01-11 12:21:53  INFO | root | Read from environment => ok'
'2025-01-11 12:21:53  INFO | root | Read from /homeassistant/tsun-proxy/config.json => n/a'
'2025-01-11 12:21:53  INFO | root | Read from /homeassistant/tsun-proxy/config.toml => n/a'
'2025-01-11 12:21:53  INFO | root | Read from /data/options.json => ok'
'2025-01-11 12:21:53  INFO | root | ******'
'2025-01-11 12:21:53  INFO | root | listen on port: 5005 for inverters'
'2025-01-11 12:21:53  INFO | root | listen on port: 10000 for inverters'
'2025-01-11 12:21:53  INFO | mqtt | start MQTT: host:core-mosquitto  port:1883  user:addons'
'2025-01-11 12:21:53  INFO | root | HTTP server listen on port: 8127'
'2025-01-11 12:21:53  INFO | mqtt | MQTT broker connection established'
'2025-01-11 12:21:53  INFO | root | Initialize proxy device on home assistant'
'2025-01-11 12:21:54  INFO | mqtt | Home-Assistant Status: offline'
'2025-01-11 12:22:02  INFO | conn | [G3P:0] Accept connection from ('192.168.200.49', 62647)'
'2025-01-11 12:22:02  INFO | conn | [G3P:0] Peer closed. for ('172.30.33.5', 10000)'
'2025-01-11 12:22:02  INFO | conn | [G3P:0] Server loop stopped for r('192.168.200.49', 62647)'
'2025-01-11 12:22:02  INFO | conn | [G3P:1] Accept connection from ('192.168.200.50', 62682)'
'2025-01-11 12:22:02  INFO | conn | [G3P:2] Accept connection from ('192.168.200.49', 62648)'
'2025-01-11 12:22:02  INFO | conn | [G3P:3] Accept connection from ('192.168.200.50', 62683)'
'2025-01-11 12:22:02  INFO | conn | [G3P:1] Peer closed. for ('172.30.33.5', 10000)'
'2025-01-11 12:22:02  INFO | conn | [G3P:1] Server loop stopped for r('192.168.200.50', 62682)'
'2025-01-11 12:27:23  INFO | conn | [G3P:4] Accept connection from ('192.168.200.50', 58066)'
```

mime24, 2025-01-11:
Hallo @frankol,

ich habe wahrscheinlich nicht den entscheidenden Tipp, jedoch fallen mir ein paar Punkte auf.

1. In deiner config ist beim zweiten WR die monitor_sn in " geschrieben. probier mal diese Anführungszeichen (") wegzulassen. 
2. In dem Log erscheint eine Meldung welche bei mir bisher nie aufgetaucht ist.
`INFO | mqtt | Home-Assistant Status: offline'`
3. die config.toml ist im Addon-Betrieb eig nicht vorgesehen. Der Verzeichnispfad im Log beschreibt den Pfad innerhalb des Containers und ist von außen nur über addon_configs (z.B. via SMB oder VS-Code) zugänglich. der dort auftauchende Ordner (c676133d_tsun-proxy) ist im Addon auf "/homeassistant/tsun-proxy/" gemounted und enthält z.B. die detaillierten Log files.

frankol, 2025-01-11:
Hi @mime24 , 
danke für den Tipp mit der config. 
Es werden automatisch Anführungszeichen gesetzt, bei beiden. Die obere Config hatte ich aus Notepad++ rauskopiert. Im Addon werden die richtig gesetzt.

Das Mapping des Logs scheint auch nicht richtig zu laufen. Bei mir wird kein tsun-proxy Ordner unter /homeassistant angelegt. Der Log-Ordner liegt einzig im addon_configs

mime24, 2025-01-11:
Hmm. 

Ordnerstruktur sollte bei dir wie folgt aussehen:

addon_configs/c676133d_tsun-proxy/logs

Da gibts keinen Ordner der Homeassistant heißt.  Es wird auch sonst kein Ordner in deinem Homeassistant erzeugt

frankol, 2025-01-11:
ah, ich dachte es wird im homeassistant ein logs Ordner angelegt, weil man den addon_config Ordner nur per CLI oder smb erreicht. Wäre leichter, wenn der Logs Ordner unter /config/tsun-proxy/logs oder /homeassistant/tsun-proxy/logs gelinkt wird. Ändert aber leider nichts am eigentlichen Problem, dass die Inverter im mqtt nicht richtig strukturiert werden

mime24, 2025-01-11:
Auf diese Ordner hat das Addon keinen Zugriff.

Zu dem eigentlichen Problem kann ich später nur mal testen wie die von dir gelisteten Optionen an den Proxy übergeben werden. Die Verarbeitung kann ich nicht beurteilen. Da kann sicherlich Stefan helfen.

s-allius, 2025-01-11:
Hallo frankol,
spontan habe ich keine Idee, warum der eine Inverter nicht richtig eingebunden wird. Ich werde mal im Code schauen, wie dies passieren könnte.
Prima, wäre der Inhalt der `/data/options.json` zu posten. Dann kann ich entsprechende Unit-Test erstellen und das Problem untersuchen

s-allius, 2025-01-11:
Prüfe mal die Seriennummer: `Y47E00000020000CE`diese ist 17 stellen lang. der Proxy erwartet eine 16-stellige und verwirft die Config. Evtl. fehlt die Seriennummer aber auch schon in der `/data/options.json`da die Länge falsch ist

mime24, 2025-01-11:
hier der Inhalt der options.json mit den angaben von @frankol 

```

{
  "inverters": [
    {
      "serial": "Y47E00000020000CE",
      "monitor_sn": 2090000186,
      "node_id": "PV-Garage-links",
      "suggested_area": "Garage",
      "modbus_polling": true,
      "client_mode.host": "192.168.200.49",
      "client_mode.port": 8899,
      "client_mode.forward": false,
      "pv1.manufacturer": "Shinefar",
      "pv1.type": "SF-M18/144550",
      "pv2.manufacturer": "Shinefar",
      "pv2.type": "SF-M18/144550"
    },
    {
      "serial": "Y47E00000200000B",
      "monitor_sn": 2090000183,
      "node_id": "PV-Garage-mitte",
      "suggested_area": "Garage",
      "modbus_polling": false,
      "client_mode.host": "192.168.200.50",
      "client_mode.port": 8899,
      "client_mode.forward": false,
      "pv1.manufacturer": "Shinefar",
      "pv1.type": "SF-M18/144550",
      "pv2.manufacturer": "Shinefar",
      "pv2.type": "SF-M18/144550"
    }
  ],
  "tsun.enabled": true,
  "solarman.enabled": true,
  "inverters.allow_all": false,
  "gen3plus.at_acl.tsun.allow": [
    "AT+Z",
    "AT+UPURL",
    "AT+SUPDATE"
  ],
  "gen3plus.at_acl.mqtt.allow": [
    "AT+"
  ]
}

```

frankol, 2025-01-11:
> Hallo frankol, prüfe mal die Seriennummer: `Y47E00000020000CE`diese ist bei dir 17 stellen lang. Der Proxy erwartet eine genau 16-stellige und verwirft die Config, wenn diese nicht stimmt.

ohh backe. Danke für den Hinweis. Ich dachte ich hatte alles mehrfach kontrolliert... War eine 0 zu viel. Danke!

Jetzt werden beide Wechselrichter richtig einsortiert
[image]

mime24, 2025-01-11:
@s-allius Lass uns die config.yaml anpassen. Sodass die Länge der Seriennummer bereits bei der Eingabe geprüft wird. 

Es müsste str(min,max) als Typ im Schema Supporten. Ich könnte das morgen testen

frankol, 2025-01-11:
klasse Idee. Denn der Wechselrichter wurde ja auch angezeigt, nur nicht richtig einsortiert. 

Jetzt fehlt nur noch das Resetten der Watt-Zahlen, wenn die Wechselrichter offline gehen. Da habe ich aber schon ein Issue zu gesehen.
Tolles Projekt!

closed 2025-01-11 by frankol