s-allius/tsun-gen3-proxy #285: Extent the configuration for Smart Meter and Batteries

s-allius, 2025-02-14:
To support devices other than inverters, we have several approaches. At the moment, the configuration in config.toml always starts with the keyword `inverters`. Now we can:
1. simply add the new device types under `inverters` as well. Disadvantage, we do not check the grammar for the individual devices, as this depends on the data entered (e.g. serial number)
2. add separate keywords like, `smart_meter`, `battery`. Then we could customize the grammar for each type specifically.

I would prefer the second approach.

@mime24 : What do you think? How can we map this for the add-on?

mime24, 2025-02-14:
separate keywords sounds defiantly more meaningful. It should be possible to adjust the config.yaml accordingly. but I'm not available for the next 8 days to create a test config.

s-allius, 2025-02-23:
I have implemented a new keyword `batteries`in this pull request: #282

closed 2025-02-23 by s-allius