s-allius/tsun-gen3-proxy #288: logging.ini not fully respected

onkelenno, 2025-02-23:
Hi.

Again, thanks for your great work.

One think I noticed: the log levels configured in logging.ini are not fully respected.
The log levels are always overwritten for all loggers based on environment LOG_LVL.
Does that make sense if I can configure it in logging.ini? I would really like to be able to reduce the output for a single logger e.g. for tracer.

https://github.com/s-allius/tsun-gen3-proxy/blob/204bc76153cc9bda290f50b2997fd1d98be3dff8/app/src/server.py#L181

Additionally there seems to be an unexpected apostrophe at the end of both format keys in logging.ini. Without them, the output of the logging messages are much cleaner.

https://github.com/s-allius/tsun-gen3-proxy/blob/204bc76153cc9bda290f50b2997fd1d98be3dff8/app/src/logging.ini#L74

PS: I'm using tsun-gen3-proxy without docker.

s-allius, 2025-02-23:
Hi onkelenno,

thanks for you hints. With the last version, we added support to install the proxy as a HA Add-On. For this I rewrote the config parser and I had in mind that some want to install the proxy on a plain linux. So it is good to hear that you run it without docker.

I will have a look if you approaches will fit to the other operation concept...

s-allius, 2025-02-23:
Regarding the LogLvl I have an approach. The original idea was to the the levels depending of the kind of docker container, which runs. So we can easily set the log level for an debug container, etc.

So I can change the behaviour, that we don't overwrite the logging.ini, if the env variable LogLvl isn't set or have an invalid value.
I think this should work for you too, right?

s-allius, 2025-02-23:
fixed with #289

closed 2025-02-23 by s-allius

onkelenno, 2025-02-23:
Sounds great. Thanks for the quick fixes.