s-allius/tsun-gen3-proxy #317: Home Assistant and MQTT on Docker Topic name problem

SqualeDev, 2025-03-22:
After installing tsun-gen3-proxy on Docker as well as routing via PiHole also on Docker, and having seen my data uploaded to HA yesterday, this morning all the sensors on the Micro Inverter TSOL-MSxx00 under HA changed status from unknown to unavailable.

I looked in MQTT Explorer at the messages passing through my broker and you can see messages passing through the topics/tsun/{controller/env/events/grid/input/inverter/other/proxy/total}/*

So I searched HA for the names of the sensors:

[image]

and I looked for them in the topics under /homeassistant, and there was nothing:

[image]

However, I have other entries without the "module_":

[image]

How can this problem be resolved?

SqualeDev, 2025-03-22:
Est-ce que ma configuration du fichier config.toml est incorrecte ?

```
##########################################################################################
##
## MQTT broker configuration
##

mqtt.host    = '192.168.0.xxx'   # URL or IP address of the mqtt broker
mqtt.port    = 1883
mqtt.user    = 'xxxxxxxxxx'
mqtt.passwd  = 'xxxxxxxxxxxxxxxxxxxxx'

##########################################################################################
##
## HOME ASSISTANT
##

ha.auto_conf_prefix = 'homeassistant'       # MQTT prefix for subscribing for homeassistant status updates
ha.discovery_prefix = 'homeassistant'       # MQTT prefix for discovery topic 
ha.entity_prefix    = 'tsun'                # MQTT topic prefix for publishing inverter values
ha.proxy_node_id    = 'proxy'               # MQTT node id, for the proxy_node_id
ha.proxy_unique_id  = 'Y47xxxxxxxxxxxxx'    # MQTT unique id, to identify a proxy instance

##########################################################################################
##
## GEN3PLUS Proxy Mode Configuration
##

solarman.enabled = false   # false: disables connecting to the tsun cloud, and avoids updates
solarman.host    = 'iot.talent-monitoring.com'
solarman.port    = 10000

##########################################################################################
###
### Inverter Definitions
###

inverters.allow_all = false   # only allow known inverters

##########################################################################################
##
## For each GEN3PLUS inverter, the serial number of the inverter must be mapped to an MQTT
## definition. To do this, the corresponding configuration block is started with
## `[Inverter.“<16-digit serial number>”]` so that all subsequent parameters are assigned
## to this inverter. Further inverter-specific parameters (e.g. polling mode, client mode)
## can be set in the configuration block
## 

[inverters."Y47xxxxxxxxxxxxx"]
monitor_sn = xxxxxxxxxx      # The GEN3PLUS "Monitoring SN:"
node_id = ''                 # MQTT replacement for inverters serial number  
suggested_area = ''          # suggested installation place for home-assistant
## modbus_polling = true        # Enable optional MODBUS polling

# if your inverter supports SSL connections you must use the client_mode. Pls, uncomment
# the next line and configure the fixed IP of your inverter
client_mode = {host = '192.168.0.yyy', port = 8899, forward = true}  

## pv1 = {type = 'RSM40-8-410M', manufacturer = 'Risen'}   # Optional, PV module descr
## pv2 = {type = 'RSM40-8-410M', manufacturer = 'Risen'}   # Optional, PV module descr
## pv3 = {type = 'RSM40-8-410M', manufacturer = 'Risen'}   # Optional, PV module descr
## pv4 = {type = 'RSM40-8-410M', manufacturer = 'Risen'}   # Optional, PV module descr

##########################################################################################
###
### If the proxy mode is configured, commands from TSUN can be sent to the inverter via
### this connection or parameters (e.g. network credentials) can be queried. Filters can
### then be configured for the AT+ commands from the TSUN Cloud so that only certain
### accesses are permitted.
###

[gen3plus.at_acl]
# filter for received commands from the internet
tsun.allow = ['AT+Z', 'AT+UPURL', 'AT+SUPDATE']
tsun.block = []
# filter for received commands from the MQTT broker
mqtt.allow = ['AT+']
mqtt.block = []
```

s-allius, 2025-03-24:
Hi SqualeDev,

I think that the problem is around you MQTT Broker, maybe HA or the proxy can't establish a connection. 

As first step check the console of the proxy. After restart you will se a message about the MQTT connection status there.
```bash
2025-03-24 19:19:52  INFO | mqtt | start MQTT: host:mqtt  port:1883  user:stefan
2025-03-24 19:19:52  INFO | root | HTTP server listen on port: 8127
2025-03-24 19:19:52  INFO | mqtt | MQTT broker connection established
```
When this is fine, than you can restart your HA, and the proxy should be informed over MQTT about that:
```bash
2025-03-24 19:21:51  INFO | mqtt | Home-Assistant Status: offline
2025-03-24 19:21:52  INFO | root | [mtr_1/] Scan info: inv:1 reg:0400
2025-03-24 19:21:53  INFO | root | [inv_3/] Scan info: inv:1 reg:0400
2025-03-24 19:22:21  INFO | mqtt | Home-Assistant Status: online
2025-03-24 19:22:21  INFO | root | Initialize proxy device on home assistant
```

SqualeDev, 2025-03-26:
Hello,

thank you for your answer.

for the MQTT connection, the proxy seems to be well connected, as well as HA (_see proxy log_) :

[image]

For the `node_id` and `suggested_area` parameters, I set them as follows:
```
[inverters."Y47E8439081E05D6"]
monitor_sn = 2086284635      # The GEN3PLUS "Monitoring SN:"
node_id = 'Sunology-2'       # MQTT replacement for inverters serial number  
suggested_area = 'Terrasse'  # suggested installation place for home-assistant
## modbus_polling = true        # Enable optional MODBUS polling
```
Apart from these 2 parameters, the code in the config.toml file has not changed.

As a result, I can see a `192.168.0.xxx/tsun/proxy` topic and a `192.168.0.xxx/tsun/Sunology-2` topic where the data is updated regularly (_as can be seen in “history” on the right of the screenshot_):

[image]

And I've also seen a `192.168.0.xxx/homeassistant/sensor/Sunology-2` topic appear, where only `config` messages are posted every minute (_as you can see on this screenshot_):

[image]

And in HA, when I filter the MQTT integration devices on `tsun`, I see a `Micro Inverter - Terrasse` device (_the word “Terrasse” doesn't exist on my configuration, so the modification has been taken into account_):

[image]

And in this device, all the data is unavailable:

[image]

When I search for the name of the `Current` sensor on Module PV1, I get this name in HA: `sensor.module_pv1_current`.

[image]

Whereas in the topic `homeassistant/sensor/Sunology-2/cur_pv1_2086284635/config`, I have this:

```
{
  "name": "Current",
  "stat_t": "tsun/Sunology-2/input",
  "dev_cla": "current",
  "stat_cla": "measurement",
  "uniq_id": "cur_pv1_2086284635",
  "val_tpl": "{{ (value_json['pv1']['Current'] | float)}}",
  "unit_of_meas": "A",
  "ic": "mdi:gauge",
  "ent_cat": "diagnostic",
  "dev": {
    "name": "Module PV1 - Terrasse",
    "sa": "Module PV1 - Terrasse",
    "via_device": "inverter_2086284635",
    "ids": [
      "input_pv1_2086284635"
    ]
  },
  "o": {
    "name": "tsun-proxy",
    "sw": "0.12.0-rel"
  }
}
```

I don't understand how HA is supposed to find the device topic.

s-allius, 2025-03-26:
Hi,
the config topic will register the topics at home assistant. At the moment I have no idea. Maybe there is something nor fitting, since you set the `node_id` latter.
Pls try to delete all of the inverter sub-devices in HA. All inputs, collator, inverter and the proxy instances. After a restart of HA, the should be added by the proxy again.

Then you have to wait. Only when a value change, it will be send to the MQTT broker

closed 2025-06-21 by s-allius