s-allius/tsun-gen3-proxy #362: Add support for TSUN MX450

FabienNDC, 2025-04-11:
Hello,

Thank you for your great work.
Could it be possible to add support for the MX450 inverter ? (serial number starting with Y00).

The web interface looks similar to the others, but I have no idea if it could be compatible with this integration.

If I can provide some info, I would be glad to help.

s-allius, 2025-04-13:
Hi,

yes we can check, if your inverter will works with the proxy.

So first, it important for me if you prefer the installation as Add-on or as a docker container. This is important, sind the description for configuration will differ.

You wrote, that you see the same web config. This means you have a GEN3PLUS inverter with the solar man protocol. First you should check. if there is a `SSL` in the firmware name. This can be a show stopper, but must not.

could you please check the firmware:

[image]

For testing you can allow incoming connections with unknown serial numbers: https://github.com/s-allius/tsun-gen3-proxy/wiki/Configuration-kword#inverter-list

You can enable this in the add-on configuration or in the config.toml file, ob the proxy.

FabienNDC, 2025-04-13:
Hi,

Thank you so much for helping.

Firmware version is LSW5_SSL_02B0_1.04

[image]

Tsun-proxy is installed as add-on to Home Assistant.

Despite the activation of inverters.allow_all option, the regex doesn't let me save the config with my inverter serial (actually I had already tried).
Is it a bug or me misunderstanding something ?

[image]

I am not sure where the config.toml file is located on disk, maybe I could force it by writting directly with an external file editor if needed.

s-allius, 2025-04-14:
Hi,

for this test you don't need to store the inverters serial number. So easily change the first three letter to `Y17`and then you can store the config.
From you screenshot I can see, that the inverter uses the same register setting like the other GEN3PLUS inverters. Tonight I will build a test version of the Add-on for easier testing.

FabienNDC, 2025-04-14:
Oh wow, looks great :

[image]

Data flows to HA. So nice.

Thanks a lot. Should I check some things, particular features or whatever ? Would be glad to contribute in return, even if it is a very humble contribution.

FabienNDC, 2025-04-14:
One more thing, probably off-topic, maybe not.

I actually own two inverters.

I changed my configuration this way : 

- serial: R1718C390815001A
  node_id: Sunology-1
  suggested_area: Jardin
  modbus_polling: false
  client_mode.host: 192.168.20.40
  client_mode.port: 8899
  client_mode.forward: true
  pv1.manufacturer: Tengsheng_G3
  pv1.type: Tengsheng_G3
- serial: R1718C390815001B
  node_id: Sunology-2
  suggested_area: Jardin
  modbus_polling: false
  client_mode.host: 192.168.20.41
  client_mode.port: 8899
  client_mode.forward: true
  pv1.manufacturer: Tengsheng_G3
  pv1.type: Tengsheng_G3

I have two controller devices in Home Assistant (yay), but they are both mapped to the same inverter. Don't understand why. When I reboot the addon, it sometimes maps to 192.168.20.40, sometimes to 192.168.20.41, but never boths.

Am I missing something (again) ?
Is it the point of the next release ?

Thanks again.

s-allius, 2025-04-14:
> Oh wow, looks great :
> 
> [image]
> 
> Data flows to HA. So nice.
> 
> Thanks a lot. Should I check some things, particular features or whatever ? Would be glad to contribute in return, even if it is a very humble contribution.

Cool, then it will be easy to support the inverter.

s-allius, 2025-04-14:
> One more thing, probably off-topic, maybe not.
>
> I actually own two inverters.
>
> I changed my configuration this way :
>
> * serial: R1718C390815001A
>   node_id: Sunology-1
>   suggested_area: Jardin [...]

Hi FabianNDC,

it will be much better if open a own task, so it is much easier to follow and help.

You problem is, that both inverter are mapped to the same controller. This will happen, if you use the same node_id for both ones. Here I think, that the minus isn't allowed in MQTT for toppics, an so both id will be the same. Please try Sunology_1 instead of Sunology-1.

s-allius, 2025-04-14:
If you activate the advanced mode in your Home Assistant (under your user setting), than you will see a dev version of the proxy in the repository.

You can install the `TSUN-Proxy (Dev)` 
in version: 0.14.0-dev-2504141949

This version should accept your serial number and then you can disable the allow.all switch again

s-allius, 2025-04-14:
If this work, than please check if the proxy register the correct number of PV modules. I assume, that you see two or four and not one.

FabienNDC, 2025-04-14:
Thank you, after fixing the node_id, stopping the addon, deleting the mqtt devices and restarting the addon it looks way better.
I will try the dev version tomorrow when the sun sets, and will provide feedback.
Thanks again !

FabienNDC, 2025-04-15:
> If you activate the advanced mode in your Home Assistant (under your user setting), than you will see a dev version of the proxy in the repository.
>
> You can install the `TSUN-Proxy (Dev)` in [...]

Thank you, it works !
From my perspective it is ok for a proper release.
If I can do/check anything let me know.

s-allius, 2025-04-15:
It would be very nice, if you can send me a trace from the inverter traffic. The proxy logs this and a snippet over 10-15 minutes will be enough. I will check the settings and will make some unit test if needed for quality assurance.

Unfortunately it is not so easy to get the trace. You must install another Addon like `SHH` or `File editor`. Then you will find the trace under `/addon_configs/c676133d_tsun-proxy-dev/logs/trace.log`.
Please don't post the log here. Better send it by email: DataType76@allius.de

s-allius, 2025-04-15:
Thanks, I got it.

s-allius, 2025-04-15:
implemented with #379