s-allius/tsun-gen3-proxy #390: Keine Verbindung mehr zur Cloud

ranfab, 2025-04-20:
Hallo,

erstmal vielen Dank für die Tolle Isee und Umsetzung.
Soweit hat bei mir alles geklappt, und ich sehe die Daten in HomeAssistant.
Nur, obwohl ich den Haken gesetzt habe (Gen3 Cloud), werden die Daten nicht mehr in der App und auf der Webseite angezeigt.
Gibt es da eine Lösung, oder geht das nicht mehr?

Gruß 

Fabien

s-allius, 2025-04-20:
Hallo Fabien,

meist liegt es an einem Problem bei der DNS Auflösung, dass der Proxy die Cloud nicht erreichen kann. Schau mal in das Log des Proxies. Du kannst auch einfach das Add-on noch mal `restarten`, dann siehst du im Log direkt die Verbbindungsaufbauten und musst nicht lange in der Vergangenheit suchen.

Gruß Stefan

ranfab, 2025-04-20:
Hallo,

Ich denke, das wird der Part sein…

2025-04-20 16:21:16 ERROR | root | resolve logger.talent-monitoring.com to 192.168.178.204, which is a private IP!
  Check your DNS settings and use a public DNS resolver!
  To prevent a possible loop, forwarding to local IP addresses is
  not supported and is deactivated for subsequent connections

Aber sollte man laut Anleitung nicht logger.talent-monitoring.com auf den MQTT weiterleiten?!?

Gruß 

Fabian

s-allius, 2025-04-21:
Hallo Fabian,

das ist soweit alles richtig eingerichtet. Daher verbinden sich deine Geräte ja auch schon mit Proxy.

Prinzipiell benötigst du aber zwei verschiedene DNS Server:
- Den AdGuard, um interne Adress Auflösungen zu bearbeiten und damit auch die Umleitung deiner Geräte zum Proxy einzurichten. Das funktioniert bei dir. Prima
- Dann müssen aber der AdGuard und der Proxy auch Adressen aus dem Internet auflösen können. Dies geht nur wenn sie einen DNS Server im Internet erreichen. Dazu muss du im Home Assistant einen festen, öffentlichen DNS-Server einrichten.

Schau mal in HA unter `Settings`-> `System`->`Network`. Dort kannst du die Netzwerk Konfiguration von Home Assistant von Automatisch auf statisch umstellen.

[image]

Du solltest dann in deiner Fritzbox noch die IP Adresse für dein HA festlegen, damit diese "feste IP" nicht doch mal vergeben wird.

---
 
Wenn HA nun einen öffentlichen DNS Server (z.B. 8.8.8.8) benutzt, dann wird für AdGuard und den Proxy wieder `logger.talent-monitoring.com`auf die öffentliche IP-Adressen auflöst.

Also:
  DNS Auflösung vom Intranet: AdGuard mit Umleitung zum Proxy 
  DNS Auflösung für HA und AddOn: öffentlicher DNS-Server ohne Umleitung zum Proxy

closed 2025-06-21 by s-allius