s-allius/tsun-gen3-proxy #420: config.example.toml file is missing

Mojtra, 2025-05-22:
Hi, I want to manage the tsun proxy via portainer. The container was created, but the config.example.toml file is missing. 
Since I have an inverter which supports SSL connections, I have to activate the client_mode. To do so I need to edit the file. Unfortunately the config.example.toml was neither created on the host nor in the container. 
The storage configuration seems to be correct, since I can see logfiles. What have I done wrong? 

Hostfiles
[image]

Containerfiles
[image]

Containervolume configuration
[image]

Env settings
[image]

let me know if you need more informaition

s-allius, 2025-05-22:
Hi Mojtra,

you are right. I delete the example file on my installation, and after restarting the proxy the file is still missing. Thanks for reporting this bug.

I came back to you, when I have a fix for this.

s-allius, 2025-05-22:
Hi Mojtra,

I made a fix. You get a container with this fix from docker.io

```
    image: docker.io/sallius/tsun-gen3-proxy:dev
```

closed 2025-05-22 by Mojtra