s-allius/tsun-gen3-proxy #438: Proxy not starting with externl MQTT-Broker

rain-er, 2025-05-30:
Hello,
Yesterday I managed to install and run the proxy with "Add-on version: 0.13.0-rel". I was very happy!
I'm running the proxy as Homeassistant addon.
I'm using an external Mosquitto MQTT-Broker, not the HA Mosquitto Addon

Log of successful operation with "0.13.0-rel":
```
-----------------------------------------------------------
 Add-on version: 0.13.0-rel
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.2  (amd64 / qemux86-64)
 Home Assistant Core: 2025.5.3
 Home Assistant Supervisor: 2025.05.3
`-----------------------------------------------------------`
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Add-on environment started
check for Home Assistant MQTT
[14:46:09] ERROR: Got unexpected response from the API: Service not enabled
MQTT not found
Start Proxyserver...
2025-05-29 14:46:11  INFO | root | Server "proxy - 0.13.0-rel" will be started
2025-05-29 14:46:11  INFO | root | current dir: /home/proxy
2025-05-29 14:46:11  INFO | root | config_path: /homeassistant/tsun-proxy/
2025-05-29 14:46:11  INFO | root | json_config: /data/options.json
2025-05-29 14:46:11  INFO | root | toml_config: None
2025-05-29 14:46:11  INFO | root | log_path:    /homeassistant/tsun-proxy/logs/
2025-05-29 14:46:11  INFO | root | log_backups: 2 days
2025-05-29 14:46:11  INFO | root | LOG_LVL    : INFO
2025-05-29 14:46:11  INFO | root | ******
2025-05-29 14:46:11  INFO | root | Read from /home/proxy/cnf/default_config.toml => ok
2025-05-29 14:46:11  INFO | root | Read from environment => ok
2025-05-29 14:46:11  INFO | root | Read from /homeassistant/tsun-proxy/config.json => n/a
2025-05-29 14:46:11  INFO | root | Read from /homeassistant/tsun-proxy/config.toml => n/a
2025-05-29 14:46:11  INFO | root | Read from /data/options.json => ok
2025-05-29 14:46:11  INFO | root | ******
2025-05-29 14:46:11  INFO | root | listen on port: 5005 for inverters
2025-05-29 14:46:11  INFO | root | listen on port: 10000 for inverters
2025-05-29 14:46:11  INFO | mqtt | start MQTT: host:192.168.178.83  port:1883  user:srv_mqtt
2025-05-29 14:46:11  INFO | root | HTTP server listen on port: 8127
2025-05-29 14:46:11  INFO | mqtt | MQTT broker connection established
2025-05-29 14:46:11  INFO | root | Initialize proxy device on home assistant
2025-05-29 14:46:12  INFO | conn | [G3:0] Accept connection from ('192.168.178.70', 49321)
```

Today early in the morning I made the update of Homeassistant Addon to "0.14.0-rel".
Now the proxy is not starting any more.
I guess it has to do with MQTT Broker. But I have no idea what could be wrong.
Do you have an Idea?

```
-----------------------------------------------------------
 Add-on: TSUN-Proxy
 MQTT Proxy for TSUN Photovoltaic Inverters
-----------------------------------------------------------
 Add-on version: 0.14.0-rel
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.2  (amd64 / qemux86-64)
 Home Assistant Core: 2025.5.3
 Home Assistant Supervisor: 2025.05.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Add-on environment started
check for Home Assistant MQTT
[04:09:22] ERROR: Got unexpected response from the API: Service not enabled
[04:09:22] ERROR: Failed to get services from Supervisor API
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

This is my confguration from Homeassistant UI:
I have a GEN3 MS600, no Batteries.

```
inverters:
  - serial: R17EXXXXXXXXX
    monitor_sn: 0
    node_id: R17EXXXXXXXXX
    suggested_area: Solar Garage
    modbus_polling: false
    pv1.manufacturer: Risen
    pv1.type: RSM40-8-405M
    pv2.manufacturer: Risen
    pv2.type: RSM40-8-405M
batteries:
  - serial: 4100000000000000
    monitor_sn: 0
    node_id: bat_1
    suggested_area: Garage
    modbus_polling: false
    pv1.manufacturer: Shinefar
    pv1.type: SF-M18/144550
    pv2.manufacturer: Shinefar
    pv2.type: SF-M18/144550
tsun.enabled: true
solarman.enabled: false
inverters.allow_all: false
gen3plus.at_acl.tsun.allow:
  - AT+Z
  - AT+UPURL
  - AT+SUPDATE
gen3plus.at_acl.mqtt.allow:
  - AT+
mqtt.host: 192.168.178.83
mqtt.port: 1883
mqtt.user: XXXXXXXXXXXX
mqtt.passwd: XXXXXXXXXXXX
```
Gruß
Rainer

s-allius, 2025-05-30:
Hi rain-er,

I also saw this problems sometimes. The proxy can't get the MQTT credentials. I have no clue why. But in my case restarting the MQTT Aden solve this. If this doesn't help please try to restart the Home Assistant.

mime24, 2025-05-30:
Thats interesting. We might have to spend some thoughts on coping with the supervisor response when checking for MQTT Addon in run.sh

mime24, 2025-05-30:
I just reviewed your log again. It's weird that the supervisor call produce this error:

`[04:09:22] ERROR: Failed to get services from Supervisor API`

It looks like a supervisor issue. Not running properly? But you restarted HA. Super weird. Maybe we need to check whether there is a change in the behavior of supervisor. Did you recognize any system/supervisor related update?

s-allius, 2025-05-31:
With the version 0.14.0 I also upgraded the add-on base image. For me it seems that the is a change so that the S6 startup breaks. I will check if there are known errors in the base-image.

s-allius, 2025-05-31:
OK, I found the problem. In bashio was a change last month in the error handling of supervisors call. The new code now returns "${__BASHIO_EXIT_NOK}".  Before the change we always get a "${__BASHIO_EXIT_OK}" even on errors. So we only check if the MQTT values are set.
I think we have to add some code to catch the error return value. Because I am not familiar with bash, I have to read a little bit for a solution...

rain-er, 2025-05-31:
Hallo Stefan,

yes, 0.14.0-dev funktioniert.

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: TSUN-Proxy (Dev)
 MQTT Proxy for TSUN Photovoltaic Inverters
-----------------------------------------------------------
 Add-on version: 0.14.0-dev-2505311918
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.2  (amd64 / qemux86-64)
 Home Assistant Core: 2025.5.3
 Home Assistant Supervisor: 2025.05.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
run.sh: info: Add-on environment started
run.sh: info: check for Home Assistant supervisor API
run.sh: info: check for Home Assistant MQTT
[21:09:41] ERROR: Got unexpected response from the API: Service not enabled
[21:09:41] ERROR: Failed to get services from Supervisor API
run.sh: error: Home Assistant service MQTT not available!
run.sh: info: MQTT configuration not found
Start Proxyserver...
2025-05-31 21:09:43  INFO | root | Server "proxy - 0.14.0-dev-2505311918" will be started
2025-05-31 21:09:43  INFO | root | current dir: /home/proxy
2025-05-31 21:09:43  INFO | root | config_path: /homeassistant/tsun-proxy/
2025-05-31 21:09:43  INFO | root | json_config: /data/options.json
2025-05-31 21:09:43  INFO | root | toml_config: None
2025-05-31 21:09:43  INFO | root | trans_path:  ../translations/
2025-05-31 21:09:43  INFO | root | rel_urls:    True
2025-05-31 21:09:43  INFO | root | log_path:    /homeassistant/tsun-proxy/logs/
2025-05-31 21:09:43  INFO | root | log_backups: 2 days
2025-05-31 21:09:43  INFO | root | LOG_LVL    : INFO
2025-05-31 21:09:43  INFO | root | ******
2025-05-31 21:09:43  INFO | root | Copy Default Config to /homeassistant/tsun-proxy/config.example.toml
2025-05-31 21:09:43  INFO | root | Read from /home/proxy/cnf/default_config.toml => ok
2025-05-31 21:09:43  INFO | root | Read from environment => ok
2025-05-31 21:09:43  INFO | root | Read from /homeassistant/tsun-proxy/config.json => n/a
2025-05-31 21:09:43  INFO | root | Read from /homeassistant/tsun-proxy/config.toml => n/a
2025-05-31 21:09:43  INFO | root | Read from /data/options.json => ok
2025-05-31 21:09:43  INFO | root | ******
2025-05-31 21:09:43  INFO | root | Start Quart
 * Serving Quart app 'server'
 * Debug mode: False
 * Please use an ASGI server (e.g. Hypercorn) directly in production
 * Running on http://0.0.0.0:8127 (CTRL + C to quit)
2025-05-31 21:09:43  INFO | root | listen on port: 5005 for inverters
2025-05-31 21:09:43  INFO | root | listen on port: 10000 for inverters
2025-05-31 21:09:43  INFO | mqtt | start MQTT: host:192.168.178.83  port:1883  user:srv_mqtt
[2025-05-31 21:09:43 +0200] [176] [INFO] Running on http://0.0.0.0:8127 (CTRL + C to quit)
2025-05-31 21:09:43  INFO | hypercorn.error | Running on http://0.0.0.0:8127 (CTRL + C to quit)
2025-05-31 21:09:43  INFO | mqtt | MQTT broker connection established
2025-05-31 21:09:43  INFO | root | Initialize proxy device on home assistant
```

s-allius, 2025-05-31:
Cool, I will revise the log outputs a little and build a new release. The fix will be available with version 0.14.1.-rel